fougue/mayo #10: compile error!

461353349, 2018-09-01:
There are two compile error. Win10, qt5.10, vs2017, occt7.3

qt_utils.obj:-1: error: LNK2019: unresolved external symbol "public: __cdecl TCollection_ExtendedString::TCollection_ExtendedString(char16_t const * const)" (??0TCollection_ExtendedString@@QEAA@QEB_S@Z) referenced in function "public: static class TCollection_ExtendedString __cdecl occ::QtUtils::toOccExtendedString(class QString const &)" (?toOccExtendedString@QtUtils@occ@@SA?AVTCollection_ExtendedString@@AEBVQString@@@Z)

qt_utils.obj:-1: error: LNK2019: unresolved external symbol "public: char16_t const * __cdecl TCollection_ExtendedString::ToExtString(void)const " (?ToExtString@TCollection_ExtendedString@@QEBAPEB_SXZ) referenced in function "public: static class QString __cdecl occ::QtUtils::toQString(class TCollection_ExtendedString const &)" (?toQString@QtUtils@occ@@SA?AVQString@@AEBVTCollection_ExtendedString@@@Z)

HuguesDelorme, 2018-09-03:
Hello,

This is a linking error. I changed recently the qmake arguments for specifying OpenCascade include path  (dir containing `*.hxx`) and OpenCascade lib path (dir containing `*.lib`).
See commit https://github.com/fougue/mayo/commit/8546008085a583a054ff00de33c0a394025fa552 and also the README "Build instructions" 
Maybe this is related. Are you trying to build the "develop" branch ?
If so then do a "git pull" and make sure to :
    `qmake "CASCADE_INC_DIR=your_occ_headers_path" "CASCADE_LIB_DIR=your_occ_libs_path"`

461353349, 2018-09-05:
It is OK, Fine, Thank you!

closed 2018-09-05 by 461353349