fougue/mayo #11: Can Selection Highlight improve?

461353349, 2018-09-22:
When mouse move, the lines of face is highlighting. If the face not lines is highlighting, It would be better.

HuguesDelorme, 2018-09-27:
Yes you are right, it's pretty annoying.
Actually this is the call to AIS_InteractiveContext::MoveTo() in mainwindow.cpp that triggers the highlight.
It can be improved a bit by passing false to the last flag "theToRedrawOnUpdate" of MoveTo().
But even with this highlight is triggered when the user over some part and wants to rotate/pan the 3D view.
Anyway, this thing needs to be sorted out.

461353349, 2018-10-07:
@HuguesDelorme Thanks for your reply.  There are some changes on highlighting in OCCT7.2, I donnot know if it is useful.
_Since Highlight and Selection styles within AIS_InteractiveContext are now defined by Prs3d_Drawer inheriting from Graphic3d_PresentationAttributes, it is now possible to customize default highlight attributes like Display Mode and ZLayer, which previously could be defined only on Object level._

closed 2018-10-20 by 461353349