fougue/mayo #13: Which version of Qt is used? Is msvc2017?

jinlong631, 2019-09-16:
Which version of Qt is used? Is msvc2017?

HuguesDelorme, 2019-09-16:
Hello!
The source code relies on Qt5, about the minimal version I can't tell for sure. On my workstation Mayo builds fine with Qt 5.11 and 5.13 but it is surely compatible with older versions.
And yes you can build with MSVC 2017 on Windows, actually this is the compiler I currently use.

jinlong631, 2019-09-17:
> Hello!
> The source code relies on Qt5, about the minimal version I can't tell for sure. On my workstation Mayo builds fine with Qt 5.11 and 5.13 but it is surely compatible with older versions.
> And yes you can build with MSVC 2017 on Windows, actually this is the compiler I currently use.

Okay thank you

closed 2019-11-04 by HuguesDelorme