fougue/mayo #15: Convert mouse position to world coordinates

sbabadag, 2020-03-15:
hi;
the V3dView_to3dPosition function works only if there is an object under the mouse cursor, but when its an empty space the coordinates are not correct?

HuguesDelorme, 2020-03-16:
Hello Selahattin,

Function `GpxUtils::V3dView_to3dPosition()` is called only when picking attempt failed.
These "3D" mouse coordinates are computed in `MainWindow::onGuiDocumentAdded()`
```cpp
        auto selector = guiDoc->aisInteractiveContext()->MainSelector();
        selector->Pick(pos2d.x(), pos2d.y(), guiDoc->v3dView());
        const gp_Pnt pos3d =
                selector->NbPicked() > 0 ?
                    selector->PickedPoint(1) :
                    GpxUtils::V3dView_to3dPosition(guiDoc->v3dView(), pos2d.x(), pos2d.y());
        m_ui->label_ValuePosX->setText(QString::number(pos3d.X(), 'f', 3));
        m_ui->label_ValuePosY->setText(QString::number(pos3d.Y(), 'f', 3));
        m_ui->label_ValuePosZ->setText(QString::number(pos3d.Z(), 'f', 3));
    });
```

`V3dView_to3dPosition()` computes a 3D position by projecting the mouse position to a plane perpendicular to the actual view direction. See the coordinates with the "TOP" view for example(Z coord should be fixed).

sbabadag, 2020-03-16:
Understand  thanks for the quick answer;
but is there a way that we can get the real coordinates in a ortho view?

HuguesDelorme, 2020-03-16:
To do this I guess the user has to define a "privileged plane" and then mouse coordinates are computed inside that fixed plane.
For now it's useless in Mayo as it doesn't provide CAD commands like sketching, right ?

sbabadag, 2020-03-17:
Thanks thats right.

closed 2020-03-17 by sbabadag