fougue/mayo #25: How to fix the running error?

softempire, 2020-10-08:
[image]

HuguesDelorme, 2020-10-08:
What is your build environment ?
Qt version ?
OpenCascade version ? Did you install pre-built binaries from OpenCascade Download Center ?

If the problem persists try to load mayo.exe in Dependency Walker utility (make sure to Qt and OpenCascade DLLs can be found)

softempire, 2020-10-08:
I'm using OpenCascade OpenCASCADE-7.4.0-vc14-64, and Visual studio is 2019
Qt is 5.13.1

softempire, 2020-10-08:
[image]

softempire, 2020-10-08:
I'm using the command line. I launched a Visual Studio dev console and then QMake and then nmake

HuguesDelorme, 2020-10-08:
OK try to update your git clone with latest changes from origin/develop
I know there is a broken commit with missing *.qm files referenced in mayo.qrc
Maybe you are experiencing this

softempire, 2020-10-08:
> OK try to update your git clone with latest changes from origin/develop
> I know there is a broken commit with missing *.qm files referenced in mayo.qrc
> Maybe you are experiencing this

Thank you, I'll try this.

HuguesDelorme, 2020-10-08:
Are you using pre-built Qt binaries ? For which VisualStudio version ?

HuguesDelorme, 2020-11-02:
Hello @softempire
Did you solve this issue ?

closed 2021-01-03 by HuguesDelorme