fougue/mayo #26: Create an AppImage for Mayo

luzpaz, 2020-10-10:
Would it be possible to do this so compiling wouldn't be necessary ?  
this is also the way FreeCAD does it.

HuguesDelorme, 2020-10-12:
Hello Luzpaz,
This is a good idea, will try to do so.
Some CI(eg travis) should be in place first so creation of the AppImage can be automated.

I leave this issue opened for now.
Thanks for that suggestion!

probonopd, 2020-12-19:
The first step would be to get this to compile on the oldest still-supported Ubuntu release, which currently is xenial. Are there step-by-step instructions for it somewhere?

HuguesDelorme, 2021-03-11:
@probonopd
I've added a [wiki page](https://github.com/fougue/mayo/wiki/Build-instructions-for-Debian) for build instructions on Debian-based systems.

For recent Qt DLLs on Xenial it looks pretty straightforward, I would just pick one from https://launchpad.net/~beineri
For OpenCascade it will have to be built from source code by hand, I think its dependencies could use Xenial official packages.

I'm wondering about GCC as Mayo requires GCC >= 9
Do you know some ppa I could use to have GCC9 packages on Xenial ?

Thanks for your help

probonopd, 2021-03-12:
> I'm wondering about GCC as Mayo requires GCC >= 9
> Do you know some ppa I could use to have GCC9 packages on Xenial?

Apparently the developers put no thought into compiling and running this software on all still-supported releases of major Linux distributions, but instead (lazily?) cared just about the latest and greatest?

In this case, https://github.com/AppImageCrafters/appimage-builder may be a way to go since it allows you to build on newer and still run on older systems. Be aware that this comes at a cost, though: Almost _everything_ needs to be bundled privately.

HuguesDelorme, 2021-03-13:
@probonopd
Thanks for you help !
My first intent is to build Mayo on Xenial as you recommended and then package the binaries into an AppImage.
GCC >= 9 is required to build Mayo on Linux (mainly caused by the 3rdparty header-only lib [magic_enum](https://github.com/Neargye/magic_enum)).
So again my question : is there a way to have GCC 9 on Xenial to build Mayo ?
Or is it non-sense to do so and then use instead appimage-builder ?

probonopd, 2021-03-13:
> is there a way to have GCC 9 on Xenial to build Mayo?

I don't know, sorry. But usually pulling in newer compilers tends to pull in newer runtime requirements, and hence makes things not runnable on Xenial anymore.

Unless you use something like appimage-builder.

HuguesDelorme, 2022-09-09:
AppImage provided starting from mayo v0.6.0

closed 2022-09-09 by HuguesDelorme

probonopd, 2022-09-10:
Thanks for making an AppImage. Any chance you could build it on the oldest still-supported Ubuntu LTS release, which currently is Ubuntu 18.04? This way it could run on all still-supported releases of Ubuntu and could pass the tests on appimage.github.io.

HuguesDelorme, 2022-09-12:
@probonopd 
I will try to build on Unbuntu 18.04 with LLVM/Clang.
`magic_enum` 3rdparty library used in Mayo requires GCC >= 9 and Ubuntu 18.04 provides GCC 7.4.0
`magic_enum` claims to support LLVM >=6 and it seems fine with Ubuntu 18.04