fougue/mayo #27: info request where to find the current loaded dwawing information.

grotius-cnc, 2020-10-27:
Hi,

Excellent program !! I am able to compile it on linux debian buster through changing some include paths.

Trough the code complexity i can't really find how to retrieve positions of current loaded drawing parts. In my case a 6 axis robot in step file format. I would suspect some data array's, structs, or vector array's somewhere. But i am unable to find them.

I want to be able to rotate the parts, and translate them. It sounds quite easy. I can make buttons and sliders already in the program side tab. The slot and signals don't work, but i can connect them manually, no problem.

[https://raw.githubusercontent.com/grotius-cnc/mayo/develop/mayo.png](url)

Just imagine your program working as RoKiSim, this is a basic robot simulator, but then connected to real hardware by the hal layer of Ethercat RTos. See my reprository. It can also be connected to linuxcnc.

Is it possible to write some tiny header file for me that can be used for this job? 
A few lines of code that are pointing out how to get the info out of the drawing would be perfect !

HuguesDelorme, 2020-10-27:
Hello Grotius,

Interesting !

Mayo uses OpenCascade's `XCAFPrs_AISObject` to display the STEP assembly tree.
For each "entity" an `AIS_InteractiveObject` is created. An "entity" being considered as a root tree node (ie `TDF_Label`) in the document container(see file `src/base/document.h`). Technically for a STEP assembly tree the "entities" are the "top-level free shapes" as obtained with `XCAFDoc_ShapeTool::GetFreeShapes()`.
So Mayo will create an `XCAFPrs_AISObject` for each "entity" of your STEP assembly tree (maybe there is just one root it depends how the assembly was created from the originating CAD software) see the function `GraphicsShapeEntityDriver::createEntity(const TDF_Label&)` which creates the 3D graphics mapping from an entity.

For your requirement, I see two solutions :
- Change Mayo source code so that it creates an `XCAFPrs_AISObject` for any "instance" node in the STEP assembly tree so you can change the 3D placement of the corresponding robot segments
- Rework your STEP file so that each robot segment is in its own "root" instance node then Mayo will create directly an `XCAFPrs_AISObject` for each and you will be able to change the 3D placement as above

Second solution is easier as you don't have to tweak Mayo's internals but you need to edit and adapt the STEP file (maybe with the help of FreeCad ?)

You can retrieve the 3D graphics associated to an "entity" with `GuiDocument::findGraphicsEntity()` and passing the assembly treenode identifier as argument. That functions returns a `GraphicsEntity` object containing the abstract `AIS_InteractiveObject` mapped to the entity. Call then `AIS_InteractiveObject::SetLocalTransformation()` in reaction of your button/slider signals to update the 3D placement of the affected robot segments.

Let me know how you deal this, thanks again for trying Mayo !

HuguesDelorme, 2020-10-27:
By the way feel free to do a pull request for your include paths fix so it can be merged in Mayo develop branch.
Thanks !

grotius-cnc, 2020-10-27:
Hi, Thanks for your fast reply !

I will document the install steps and report back. Have some problems now after installing another occ lib from source.
Have to do a clean os install for this and then install the occ libs by 
$ sudo apt-get install liboce-* 

If i used an older debian version, like version 9, i could made the appimage for you. But for now that is not supported.

We go for solution 2, with the help of freecad.

I have to study all your named suggestions. To get more experence with occ functions, i also play with https://github.com/Felipeasg/QtOccMapShapes.

Mayo looks so professional. Wow !

grotius-cnc, 2020-10-28:
Hi, this message is for your info. In this message I report my install steps from a fresh os and qt install  : 

1. git clone https://github.com/fougue/mayo.git

2. install opencascade library with $ sudo apt-get install ..

libocct-data-exchange-7.3
libocct-data-exchange-dev
libocct-doc
libocct-draw-7.3
libocct-draw-dev
libocct-foundation-7.3
libocct-foundation-dev
libocct-modeling-algorithms-7.3
libocct-modeling-algorithms-dev
libocct-modeling-data-7.3
libocct-modeling-data-dev
libocct-ocaf-7.3
libocct-ocaf-dev
libocct-visualization-7.3
libocct-visualization-dev
occt-draw
occt-misc

Comment about install library.

	If you install the libs at once by $ sudo apt-get install liboce-*
	you will end up with 100+ errors.
	It will install files in : /usr/share/occ/

	In the way above, by separate install, it will install in /usr/shar/opencascade/
	And the correct include path in /usr/include.. instead of /usr/share/..

3. Add include path for opencascade library in mayo.pro file 

In Mayo.pro file line 45 
+ /usr/include/opencascade/

4. To solve :

file : io_occ_vrml.ccp, 
error : no member named 'ConvertDocument' in 'VrmlData_ShapeConvert'
line 58 commented out  =>> //data.graphics->SetCappingColor(fnGetCappingColor(data));

file : widget_clip_planes.cpp, 
error : No member named "SetCappingColor"
line 70 commented out. ==>  //data.graphics->SetCappingColor(fnGetCappingColor(data));

file : Widget_clip_planes.cpp 
error : Too many arguments, have 3, takes only 1.
line 255, changed 

	from : imageCapping->Load(fileContentsData, fileContents.size(), filenameUtf8.constData())
 	==> to : imageCapping->Load(fileContents.size());

5. Mayo run's oke ! 
message : "opencascade.conf" doesn't exist or is not readable

HuguesDelorme, 2020-10-28:
Thanks for these infos, actually the build is currently broken for OpenCascade 7.3.0 with the recent changes I committed.
I will fix that asap and let you know when it's done !

HuguesDelorme, 2020-10-28:
1. OK

2. OK
Notice that Mayo will certainly not work with OCE(OpenCascade Community Edition)
Minimal supported version is OpenCascade 7.3.0

3. Instead of editing `mayo.pro` you can call qmake from the command line(or in QtCreator) like this :
`$> qmake CASCADE_INC_DIR=/usr/include/opencascade mayo.pro`
Have a look at the "build instructions" in the Mayo's README page

4. The compilation errors you reported should be solved with the last changes committed in the the develop branch(see commit 15955c9dca7ea3c9026076f84acf86257eda536c)
Please try with HEAD version of the develop branch and let me know about any problem

5. You can safely ignore this message

grotius-cnc, 2020-10-28:
Hi, Oke i have perfect news. It works without problems now.

grotius-cnc, 2020-10-28:
I solved it. I took quite a while, but after all, if i do a git clone now, it compiles and run without any problem.

I have modified the .pro file at the end a little bit. Take a look : https://github.com/grotius-cnc/mayo/blob/develop/mayo.pro

grotius-cnc, 2020-10-30:
Hi Hugues,

I was able to rotate the kuka robot parts. I studied the opencascade code and examples for a while 
to get this going.

Here is the actual qt project in zip format :
https://github.com/grotius-cnc/JellyCAD/releases/tag/1.0.0
It has some buttons and a slider to test rotating. 

There is a (big) problem with the speed of rotating. It's very slow. 
About every second a rotation matrix is done. Do you have any idea's to get this rotating speed, 
interacting as a unreal game engine? 

Is it possible to purge the actual shape into a lightweight 3d face to increase speed?
I have no idea at the moment.. Maybe you have some idea's that i can investegate?

Thanks a lot !!

HuguesDelorme, 2020-10-30:
Rotation is slow because each time you transform the BRep shape itself and call AIS_Shape::SetShape() which will recreate complete 3D presentations.
Instead of applying a transformation to the BRep shape itself apply it instead to the AIS_Shape object with AIS_Shape::SetLocalTransformation()

grotius-cnc, 2020-10-30:
Hi Hugues,

Yes, that worked !! Many thanks for the tip, you are awesome !!

The robot rotation is super fast now !! This is perfect.
'One small step for you, one giant step for me'

Working code for info :  https://github.com/grotius-cnc/JellyCAD/releases/download/1.0.1/JellyCAD_robot_rotating_oke.zip

Variables :
`TopoDS_Shape aShape;
Handle(AIS_Shape) displayShape;`

Load the drawing code :
`    STEPControl_Reader aReader;
    aReader.ReadFile("robot_in_parts.step");
    std::cout << aReader.TransferRoots() << " roots transferred." << std::endl;
    aShape  = aReader.OneShape();
    displayShape = new AIS_Shape(aShape);
    m_context->Display(displayShape,0);`

The slider code :
`   gp_Trsf myTrsf;
    myTrsf.SetRotation(gp_Ax1(gp_Pnt(0,0,0),gp_Dir(0,1,0)),value * M_PI /180);
    displayShape->SetLocalTransformation(myTrsf);
    m_context->Display (displayShape, Standard_False);
    m_context->CurrentViewer()->Redraw();`

**mes condoléances pour les pertes, nous continuerons de lutter pour la liberté d'expression!**

HuguesDelorme, 2020-10-30:
Good news you could achieve what you want.
If you're happy then please give a start to Mayo !

closed 2020-11-02 by HuguesDelorme