fougue/mayo #29: Little question about displaying the step file colors with opencascade

grotius-cnc, 2020-11-02:
Hi,

The robot program is now working and tested oke realtime with motors.
[https://github.com/grotius-cnc/Skynet_Robot_Control_Rtos_Ethercat](url)

Can you please point me in the right direction? 
My last thing to solve is to display the step file colors as in mayo. After many hours i did not succeed. I thought maybe i can ask over here. Excuse me if you see this as polution of your repro.

Here is a piece of code that displays the robot. Maybe you can look at it, and have an idea or example to display the colors. Thanks a lot !

`    Handle(TDocStd_Document) doc;
    
    STEPCAFControl_Reader aReader; //https://old.opencascade.com/content/getting-part-properties-form-iges-file
    aReader.SetColorMode(true);
    aReader.SetNameMode(true);
    aReader.SetLayerMode(true);
    aReader.ReadFile("KR6_KR10_R900.stp");
    doc = new TDocStd_Document("StepReader");
    aReader.Transfer(doc);
    
    // Assembly = XCAFDoc_DocumentTool::ShapeTool(doc->Main());
    
    Handle(XCAFDoc_ShapeTool) Assembly = XCAFDoc_DocumentTool::ShapeTool(doc->Main());
    TDF_LabelSequence frshapes;
    Assembly->GetShapes(frshapes);
    TopoDS_Shape shape = Assembly->GetShape(frshapes.Value(1));
    Handle(AIS_Shape) ais_shape = new AIS_Shape(shape);
    
    //get a label of shape.
    TDF_Label aLabel;
    aLabel = Assembly->FindShape(shape);
    
    Handle(XCAFDoc_ColorTool) myColors = XCAFDoc_DocumentTool::ColorTool(doc->Main());
    
    //TDF_LabelSequence ColLabels;
    //myColors->GetColors(ColLabels);
    
    Quantity_Color col;
    XCAFDoc_ColorType ctype;
    myColors->GetColor(aLabel,ctype,col);
    
    ais_shape->SetColor(col);
    
    m_context->Display(ais_shape,0);
    m_view->Redraw();`

HuguesDelorme, 2020-11-02:
Hello Grotius,
Yes sorry I will have to close this one, I invite you to post your question on the OpenCascade "Usage issues" forum ;
https://old.opencascade.com/forums

See you there !

closed 2020-11-02 by HuguesDelorme

grotius-cnc, 2020-11-02:
Thank you, See you there !