fougue/mayo #30: Conversion tool by CLI

HuguesDelorme, 2020-11-05:
Provide conversion tool through CLI(Command Line Interface).
Possibly something like :
```bash
$> mayo file.step --export file.obj file.stl
```

VictorLamoine, 2020-11-14:
Adding a precision switch (when applicable) would be great too. When exporting to STL I like to be able to choose the precision because it impacts 3D print quality on round features.

Or export in AMF which is a much smarter format :smile:

HuguesDelorme, 2021-05-17:
@VictorLamoine 
> Adding a precision switch (when applicable) would be great too. When exporting to STL I like to be able to choose the precision because it impacts 3D print quality on round features.

Hello Victor, have a look at develop branch and the new feature about "meshing of BRep shapes"
Relates to https://github.com/fougue/mayo/issues/56 and https://github.com/fougue/mayo/issues/61 (show the amount of triangles)

VictorLamoine, 2021-05-27:
I just tried the latest develop branch but I still experience https://github.com/fougue/mayo/issues/48 and have not time to try and fix this issue  at the moment sorry !

closed 2021-06-02 by HuguesDelorme

VictorLamoine, 2021-06-02:
I just installed OpenCascade from `apt` and used it instead my own compiled version and everything now works.
Thanks for the command line tool it works great and is fast! 👍🏼