fougue/mayo #31: Compile error

VictorLamoine, 2020-11-08:
```bash
$ lsb_release -a
No LSB modules are available.
Distributor ID:	Ubuntu
Description:	Ubuntu 20.04.1 LTS
Release:	20.04
Codename:	focal
```

```
$ g++ --version
g++ (Ubuntu 9.3.0-17ubuntu1~20.04) 9.3.0
```

`opencascade-7.4.0` compiled + installed from source

```bash
mkdir -p $HOME/software/mayo/build
cd $HOME/software/mayo/
git clone https://github.com/fougue/mayo.git src
cd build
export CSF_OCCTIncludePath=/usr/local/include/opencascade/
export CSF_OCCTLibPath=/usr/local/lib/
qmake ../src
make -j4
```

Compile error:
```bash
$ make -j12
g++ -c -pipe -std=c++17 -O2 -std=gnu++1z -Wall -W -D_REENTRANT -fPIC -DHAVE_CONFIG_H -DHAVE_FSTREAM -DHAVE_IOSTREAM -DHAVE_IOMANIP -DHAVE_LIMITS_H -DOCCT_HANDLE_NOCAST -DLIN -DLININTEL -DOCC_CONVERT_SIGNALS -D_OCC64 -DQT_NO_DEBUG -DQT_WIDGETS_LIB -DQT_GUI_LIB -DQT_CORE_LIB -I../src -I. -Isrc/app -I../src/src/app -I../src/src/3rdparty -isystem /usr/local/include/opencascade -isystem /usr/include/x86_64-linux-gnu/qt5 -isystem /usr/include/x86_64-linux-gnu/qt5/QtWidgets -isystem /usr/include/x86_64-linux-gnu/qt5/QtGui -isystem /usr/include/x86_64-linux-gnu/qt5/QtCore -I. -I. -I/usr/lib/x86_64-linux-gnu/qt5/mkspecs/linux-g++ -o io_occ_step.o ../src/src/base/io_occ_step.cpp
../src/src/base/io_occ_step.cpp:66:50: error: call to ‘static QString Mayo::IO::OccStepReader::Properties::tr(const char*, const char*, int)’ uses the default argument for parameter 2, which is not yet defined
   66 |              "`life_cycle_stage` set to `design`") },
      |                                                  ^
../src/src/base/io_occ_step.cpp:69:52: error: call to ‘static QString Mayo::IO::OccStepReader::Properties::tr(const char*, const char*, int)’ uses the default argument for parameter 2, which is not yet defined
   69 |              "`life_cycle_stage` set to `analysis`") },
      |                                                    ^
../src/src/base/io_occ_step.cpp:70:82: error: call to ‘static QString Mayo::IO::OccStepReader::Properties::tr(const char*, const char*, int)’ uses the default argument for parameter 2, which is not yet defined
   70 |         { int(ProductContext::Both), textId("Both"), tr("Translates all products") }
      |                                                                                  ^
../src/src/base/io_occ_step.cpp:71:5: error: could not convert ‘{{((int)Mayo::IO::OccStepReader::ProductContext::Design), Mayo::IO::OccStepReader::Properties::textId(const char*)(), <expression error>}, {((int)Mayo::IO::OccStepReader::ProductContext::Analysis), Mayo::IO::OccStepReader::Properties::textId(const char*)(), <expression error>}, {((int)Mayo::IO::OccStepReader::ProductContext::Both), Mayo::IO::OccStepReader::Properties::textId(const char*)(), <expression error>}}’ from ‘<brace-enclosed initializer list>’ to ‘const Mayo::Enumeration’
   71 |     };
      |     ^
      |     |
      |     <brace-enclosed initializer list>
../src/src/base/io_occ_step.cpp:76:41: error: call to ‘static QString Mayo::IO::OccStepReader::Properties::tr(const char*, const char*, int)’ uses the default argument for parameter 2, which is not yet defined
   76 |              "(not with sub-assemblies)") },
      |                                         ^
../src/src/base/io_occ_step.cpp:80:79: error: call to ‘static QString Mayo::IO::OccStepReader::Properties::tr(const char*, const char*, int)’ uses the default argument for parameter 2, which is not yet defined
   80 |              "applications requiring specialized processing of assembly parts") },
      |                                                                               ^
../src/src/base/io_occ_step.cpp:84:71: error: call to ‘static QString Mayo::IO::OccStepReader::Properties::tr(const char*, const char*, int)’ uses the default argument for parameter 2, which is not yet defined
   84 |              "with specific product, as a complement to assembly mode") },
      |                                                                       ^
../src/src/base/io_occ_step.cpp:88:65: error: call to ‘static QString Mayo::IO::OccStepReader::Properties::tr(const char*, const char*, int)’ uses the default argument for parameter 2, which is not yet defined
   88 |              "all of them are read and put in a single compound") }
      |                                                                 ^
../src/src/base/io_occ_step.cpp:89:5: error: could not convert ‘{{((int)Mayo::IO::OccStepReader::AssemblyLevel::Assembly), Mayo::IO::OccStepReader::Properties::textId(const char*)(), <expression error>}, {((int)Mayo::IO::OccStepReader::AssemblyLevel::Structure), Mayo::IO::OccStepReader::Properties::textId(const char*)(), <expression error>}, {((int)Mayo::IO::OccStepReader::AssemblyLevel::Shape), Mayo::IO::OccStepReader::Properties::textId(const char*)(), <expression error>}, {((int)Mayo::IO::OccStepReader::AssemblyLevel::All), Mayo::IO::OccStepReader::Properties::textId(const char*)(), <expression error>}}’ from ‘<brace-enclosed initializer list>’ to ‘const Mayo::Enumeration’
   89 |     };
      |     ^
      |     |
      |     <brace-enclosed initializer list>
../src/src/base/io_occ_step.cpp:93:59: error: call to ‘static QString Mayo::IO::OccStepReader::Properties::tr(const char*, const char*, int)’ uses the default argument for parameter 2, which is not yet defined
   93 |           tr("Prefer `ADVANCED_BREP_SHAPE_REPRESENTATION`") },
      |                                                           ^
../src/src/base/io_occ_step.cpp:95:62: error: call to ‘static QString Mayo::IO::OccStepReader::Properties::tr(const char*, const char*, int)’ uses the default argument for parameter 2, which is not yet defined
   95 |           tr("Prefer `MANIFOLD_SURFACE_SHAPE_REPRESENTATION`") },
      |                                                              ^
../src/src/base/io_occ_step.cpp:97:75: error: call to ‘static QString Mayo::IO::OccStepReader::Properties::tr(const char*, const char*, int)’ uses the default argument for parameter 2, which is not yet defined
   97 |           tr("Prefer `GEOMETRICALLY_BOUNDED_SURFACE_SHAPE_REPRESENTATION`") },
      |                                                                           ^
../src/src/base/io_occ_step.cpp:99:59: error: call to ‘static QString Mayo::IO::OccStepReader::Properties::tr(const char*, const char*, int)’ uses the default argument for parameter 2, which is not yet defined
   99 |           tr("Prefer `FACETTED_BREP_SHAPE_REPRESENTATION`") },
      |                                                           ^
../src/src/base/io_occ_step.cpp:101:66: error: call to ‘static QString Mayo::IO::OccStepReader::Properties::tr(const char*, const char*, int)’ uses the default argument for parameter 2, which is not yet defined
  101 |           tr("Prefer `EDGE_BASED_WIREFRAME_SHAPE_REPRESENTATION`") },
      |                                                                  ^
../src/src/base/io_occ_step.cpp:103:77: error: call to ‘static QString Mayo::IO::OccStepReader::Properties::tr(const char*, const char*, int)’ uses the default argument for parameter 2, which is not yet defined
  103 |           tr("Prefer `GEOMETRICALLY_BOUNDED_WIREFRAME_SHAPE_REPRESENTATION`") },
      |                                                                             ^
../src/src/base/io_occ_step.cpp:105:81: error: call to ‘static QString Mayo::IO::OccStepReader::Properties::tr(const char*, const char*, int)’ uses the default argument for parameter 2, which is not yet defined
  105 |           tr("Translate all representations (if more than one, put in compound)") },
      |                                                                                 ^
../src/src/base/io_occ_step.cpp:106:5: error: could not convert ‘{{((int)Mayo::IO::OccStepReader::ShapeRepresentation::AdvancedBRep), Mayo::IO::OccStepReader::Properties::textId(const char*)(), <expression error>}, {((int)Mayo::IO::OccStepReader::ShapeRepresentation::ManifoldSurface), Mayo::IO::OccStepReader::Properties::textId(const char*)(), <expression error>}, {((int)Mayo::IO::OccStepReader::ShapeRepresentation::GeometricallyBoundedSurface), Mayo::IO::OccStepReader::Properties::textId(const char*)(), <expression error>}, {((int)Mayo::IO::OccStepReader::ShapeRepresentation::FacettedBRep), Mayo::IO::OccStepReader::Properties::textId(const char*)(), <expression error>}, {((int)Mayo::IO::OccStepReader::ShapeRepresentation::EdgeBasedWireframe), Mayo::IO::OccStepReader::Proper [... cut ...] tepReader::ShapeRepresentation::GeometricallyBoundedWireframe), Mayo::IO::OccStepReader::Properties::textId(const char*)(), <expression error>}, {((int)Mayo::IO::OccStepReader::ShapeRepresentation::All), Mayo::IO::OccStepReader::Properties::textId(const char*)(), <expression error>}}’ from ‘<brace-enclosed initializer list>’ to ‘const Mayo::Enumeration’
  106 |     };
      |     ^
      |     |
      |     <brace-enclosed initializer list>
make: *** [Makefile:1697 : io_occ_step.o] Erreur 1

```

VictorLamoine, 2020-11-08:
Just realized that there is a master branch that compiles fine :+1:

HuguesDelorme, 2020-11-08:
Hello Victor,

Thanks for reporting these GCC compilation errors. I fixed them and pushed the changes in the `develop`  branch(see commit 5ade7212574529987f016dbd0244599508cc545b)
Please try again on your side and let me know if it's also OK with your build environment, so I can close this issue.

Note: there's already CI for Windows/VisualStudio2017 but CI for Linux/GCC would be also nice to have ...

VictorLamoine, 2020-11-09:
Compiles fine now!

I know 2 options:
- Travis CI which I have not used in years
- GitLab CI which I know pretty well. But you would have to mirror this repo if you don't pay for GitLab (https://docs.gitlab.com/ee/ci/ci_cd_for_external_repos/github_integration.html)

closed 2020-11-09 by VictorLamoine