fougue/mayo #35: UTF-8 part name

VictorLamoine, 2020-11-14:
Not sure if related to OpenCascade or Mayo but the part name displayed in Mayo does not support UTF-8 while the document name does:
[image]

I have noticed  OpenCascade 7.5.0 brings some UTF-8 improvements but for writing files.

HuguesDelorme, 2020-11-15:
Hello Victor, thanks for reporting this issue. I could not reproduce the issue by manually editing(ie with a text editor) an instance name within a STEP file. Can you attach yours ?
Thanks !

VictorLamoine, 2020-11-16:
Here is the example file: https://gofile.io/d/J3JsyD

HuguesDelorme, 2020-11-16:
OK looking the STEP file reaveals that the name is really stored as "Petit \X\E9tui"
Have a look at line L7600 for example :
#7025=MANIFOLD_SOLID_BREP('Petit \X\E9tui',#7016);

I guess the encoding problem here comes from the software that generated this file(steptools from the header).
Same behavior in FreeCAD 0.18 as well.

VictorLamoine, 2020-11-16:
Ah! I'm using OnShape I'll report the problem on their side then. I should have checked the file first.

closed 2020-11-16 by VictorLamoine