fougue/mayo #36: Mayo software run

TzabarDolev, 2020-11-16:
Hi, I'm not sure i fully understood, how do i enter the actual program.
I've installed Mayo, OpenCascade and i have Qt.
What is the actual run program? do i need to run something from the Qt or from console?

HuguesDelorme, 2020-11-16:
Hello, did you build Mayo from source code ?
You will need OpenCascade v7.4.0 (I'm currently porting Mayo to v7.5.0 see branch `occ/port7.5`)
If you are on Windows then use Visual Studio 2017 or 2019 to build Mayo (never tried MinGW but that should work).
Feel free to ask more help if you are experiencing problems.

Note: there is also a Windows installer for [v0.2.0](https://github.com/fougue/mayo/releases/download/v0.2.0/mayo_v0.2.0.281-3be65fe_winx64_installer.exe) but you will lack some recent features

TzabarDolev, 2020-11-16:
Hi, thanks for the fast request and sorry for the lack of information.
I'm using OpenCascade v7.5.0, ubuntu 20.04.1.
I think I've built it succesfully. The thing is, I'm not sure how to physically run the program, what is the running file?

HuguesDelorme, 2020-11-16:
OK if build is successful then look for an executable file called "mayo" from within your build directory

closed 2021-01-03 by HuguesDelorme