fougue/mayo #37: Build Error

SudatiSimone, 2020-11-25:
Hi,
What is the "(n)make" command? 
It's a simple "make"?  I run "make" (mingw32-make.exe) and there are errors due to Tab spaces. 
"Makefile.Release:516: *** missing separator.  Stop. " so I fix it by adding a tab.
But there is a new error and I don't know how to fix it. "*** [Makefile.Release:514: {src\base}.cpp{release\}.obj] Error 1"

Thanks

SudatiSimone, 2020-11-25:
Maybe I run the qmake command in the wrong way. 
This is my command:
qmake "CASCADE_INC_DIR=C:\Users\susim\Desktop\Tirocinio\OpenCascadeTechnology_OCCT\opencascade-7.5.0\inc" "CASCADE_LIB_DIR=C:\Users\susim\Desktop\Tirocinio\OpenCascadeTechnology_OCCT\opencascade-7.5.0\win64\vc14\lib"

Are there the correct folders in opencascade?

HuguesDelorme, 2020-11-25:
Hello,
"(n)make" is a name to refer to the make utility the developer is using. It should `nmake` with Visual C++ and `make` for UNIX-like toolchains.
Maybe there's a problem with the compiler identification of qmake. Can give more details about your build environment ? What compiler are you using, version of Qt and OpenCascade.

HuguesDelorme, 2020-11-25:
Have a look at [appveyor.yml](https://github.com/fougue/mayo/blob/develop/appveyor.yml) file, it might help figure out how to build.

SudatiSimone, 2020-11-25:
- I'm on Windows 10. 
- Open cascade 7.5.0
- qt5.11.2-vc14-64  (that it is inside open cascade technology) --> qmake
- Microsoft Visual Studio 2019 --> nmake
(C:\Program Files (x86)\Microsoft Visual Studio\2019\Community\VC\Tools\MSVC\14.27.29110\bin\Hostx86\x86\nmake.exe)
- MingGw --> make
"C:\Program Files\mingw-w64\x86_64-8.1.0-posix-seh-rt_v6-rev0\mingw64\bin\mingw32-make.exe"

SudatiSimone, 2020-11-25:
Now build correctly. 
There is an error during run 
[image]

HuguesDelorme, 2020-11-25:
Hello,
that's clever to reuse Qt binaries packaged with OpenCascade (then you don't need to get Qt installer).
I tried your approach and reproduced the problem.
After qmake type "nmake release"
Then in `...\OpenCascadeTechnology_OCCT\qt5.11.2-vc14-64` copy the folders `plugins\imageformats` and `plugins\platforms` and paste them into the folder where is located `mayo.exe`
You should have :
````
...\release\
        mayo.exe
        platforms\
                DLL files
        imageformats\
                DLL files        
````

Finally you should run Mayo without further problems !

PS: The error you got may be caused by the fact that the Qt packaged with OpenCascade is not relocatable (see this [doc](https://doc.qt.io/qt-5/windows-deployment.html#:~:text=For%20Windows%2C%20the%20name%20of%20the%20platform%20plugin,uses%20to%20find%20its%20plugins%2C%20as%20described%20below.))

SudatiSimone, 2020-11-26:
Uff, I don't know why but now on build there is this error: "LINK : fatal error LNK1104: cannot open file 'C:\Users\susim\Desktop\Tirocinio\OpenCascadeTechnology_OCCT\qt5.11.2-vc14-64\lib\Qt5WinExtrasd.lib'".
This file doesn't exist, there is one similar in this repository named "Qt5WinExtras.lib" so without "d"

HuguesDelorme, 2020-11-26:
Yes OpenCascade seems to have forgotten to package this `Qt5WinExtrasd.lib` file, so with the Qt bundled in OpenCascade avoid debug builds and stick to release builds instead : `nmake release`
Or preferrably install Qt5 latest version if you want also debug builds (see https://www.qt.io/download-qt-installer)

SudatiSimone, 2020-11-26:
It was my fault now it works
What a great 3d visualizer! It is very similar to cad assistant
The only feature is missing is to color the objects 

Thanks

closed 2020-11-26 by SudatiSimone