fougue/mayo #39: MinGW 8.1.0 not supported

githubuser0xFFFF, 2020-12-02:
Hi,

I just failed to compile mayo with the current MinGW 64 release 8.1.0.

Mayo uses the magic enum library. This library does not support the current MinGW release 8.1.0. The MinGW version 8.1.0 is also the version that ships with Qt. Because of this problem, it is not possible to build mayo with the current MinGW release.

HuguesDelorme, 2020-12-02:
Hello, would it be possible to keep the Qt MinGW v8 pre-built binaries and use MinGW >= 9 when building Mayo ?
Not sure how tolerant MinGW is with mixing up different versions

githubuser0xFFFF, 2020-12-02:
Thank you for your response.

Yes, I think this would be possible. The challenge is to find a pre compiled MinGW 64 compiler with the right configuration: GCC > 9.x, Posix Threads, SEH exception handling. I have not found this so far. The only option is to switch to MSYS2 MinGW but this in turn would not work with the default MinGW Qt compiler.

closed 2020-12-02 by githubuser0xFFFF