fougue/mayo #40: How debug the application

SudatiSimone, 2020-12-02:
Hi,
which are the steps to debug the application?
Ide: Microsoft visual studio 2017
I click on "Open a project or solution" and...
I have to select a file from these formats .sln, .dsw , .vcv
Where I can found it?

Thanks

HuguesDelorme, 2020-12-02:
Hello, I see two options :

1. Use QtCreator IDE where you can load `mayo.pro` then build&debug within
2. Generate a Visual Studio project (ie `.vcxproj` file) by running `qmake -t vcapp mayo.pro` from the command line. Then open that file in Visual Studio, build and debug

SudatiSimone, 2020-12-05:
Perfect. 
I'm installing qt open source for windows.
Could you tell me which components in "Select components" I have to select? above all for the version

SudatiSimone, 2020-12-05:
Solved

closed 2020-12-05 by SudatiSimone