fougue/mayo #41: Build on macOS Big Sur fails unless QMAKE_MACOSX_DEPLOYMENT_TARGET = 10.14

mschollmeier, 2020-12-09:
Without setting the MacOS version equal to or higher than 10.14, the build fails because certain `std::variant` functionalities were not available before 10.14. 
Specifically, in `occ_static_variables_rollback.cpp`, the `std::get<T>(record.value)` in lines 97, 99, 101 throws an error message " occ_static_variables_rollback.cpp:97:71: error: 'get<int, int, double, std::__1::basic_string<char> >' is unavailable: introduced in macOS 10.14
variant:1394:22: note: 'get<int, int, double, std::__1::basic_string<char> >' has been explicitly marked unavailable here" 

This can be fixed by updating `mayo.pro` with 
```
macx {
    QMAKE_MACOSX_DEPLOYMENT_TARGET = 10.14
}
```

HuguesDelorme, 2020-12-09:
Hello Marius,

    Thanks for reporting this, I never tried to build on MacOS so your feedback is very valuable.
    I have a question : does Mayo build and run fine on MacOS with your fix ?
    If so then I would love to see a pull request from you I can merge into Mayo.

Greetings!

mschollmeier, 2020-12-09:
Yes, it works just fine under MacOS, using Qt Creator and the latest Qt 5.x to build. 
I'll fork it and prepare a pull request.

mschollmeier, 2020-12-10:
Took longer than expected because my local copy was a few months behind. I've created a pull request here: #42. 

Apparently, you removed `-lTKXSDRAW` from `mayo.pro` in commit [9bcf9d2](9bcf9d2), which breaks Mayo for MacOS. Without this lib the app still compiles successfully but goes into an infinite loop that starts consuming all system RAM when executed. It works fine now, using the latest dev version of Mayo, MacOS 11.0.1 (Big Sur), Xcode 11.7, Qt 5.14.2. 

In anybody else reads this, Xcode 12 is not officially supported by Qt at the moment of this writing.

mschollmeier, 2020-12-10:
[image]

HuguesDelorme, 2020-12-10:
That's great!
I'm surprised the QWidget/OpenCascade bridge works "as-is" for macOS
Many thanks for the PR and that screenshot!

mschollmeier, 2020-12-10:
You're welcome. Thank you for creating Mayo!

closed 2020-12-10 by mschollmeier