fougue/mayo #43: Add new features: color a single component

SudatiSimone, 2020-12-11:
Hi, 
I'm trying to add a new feature to the program. I would like to color a single selected item with the color I prefer.
Above all, I add this line:
```
 m_propertyReferredColor.setUserReadOnly(false);
```
In this way, I could choose the color.

[image]

[image]

[image]

Now I want to ask you this great favor. 
1) It is possible to change also the component's color in response to the changes in the properties I did?
2) It is possible to make this change permanent? Because If I selected another component this change (the color I choose) disappears.

HuguesDelorme, 2020-12-11:
Hello,
Yes this is possible, Mayo lets you edit the Name and Product Name properties, see function `XCaf_DocumentTreeNodePropertiesProvider::Properties::onPropertyChanged()` in file `src/app/document_tree_node_properties_providers.cpp` :
````c++
    void onPropertyChanged(Property* prop) override
    {
        if (prop == &m_propertyName)
            CafUtils::setLabelAttrStdName(m_label, m_propertyName.value());
        else if (prop == &m_propertyReferredName)
            CafUtils::setLabelAttrStdName(m_labelReferred, m_propertyReferredName.value());

        PropertyGroupSignals::onPropertyChanged(prop);
    }
````
I would add another `else if (prop == &m_propertyColor)` condition and assign the color.
The thing that needs a new mechanism is to update the 3D graphics of the entity. This needs a bit of thinking!
I will have a look in the next couple of days and let you know.

SudatiSimone, 2020-12-11:
Thank you very much
I will be waiting for your reply

SudatiSimone, 2020-12-15:
I try in this way but it gives me an error for the function's scope because the path is 
Mayo::(anonymous namespace)::propertyValueText(PropertyOccColor* prop) 
and It isn't accessible from other classes for the anonymous namespace.
In document_tree_node_properties_providers.cpp,  I add:
```cpp
 #include "../app/property_item_delegate.h"
```
and 
```cpp
   void onPropertyChanged(Property* prop) override
    {
        if (prop == &m_propertyName)
            CafUtils::setLabelAttrStdName(m_label, m_propertyName.value());
        else if (prop == &m_propertyReferredName)
            CafUtils::setLabelAttrStdName(m_labelReferred, m_propertyReferredName.value());
       else if (prop == &m_propertyReferredColor){
            QString new_color = Mayo::propertyValueText(m_propertyReferredColor);
            CafUtils::setLabelAttrStdName(m_labelReferred,new_color);
        }
        PropertyGroupSignals::onPropertyChanged(prop);
    }
```
  to use this function 
```cpp
static QString propertyValueText(const PropertyOccColor* prop) {
    return StringUtils::text(prop->value());
}
```

HuguesDelorme, 2020-12-17:
Hello,
    Have a look at branch `feature/color` it implements the functionnality you described.
    There is a limitation though, as I described in some comments within the code changes.
    The supported color change applies only on the "product" shapes and not on the "instance" shapes.
    Mayo uses XCAFPrs_AISObject to display shape assemblies and this class put all shapes sharing the same "aspect style(including color)" into a shape compound. Supporting with XCAFPrs_AISObject the color change of an "instance" shape would imply to recompute whole 3D graphics of the assembly.
    I can do it also, that would interesting to see the impact.

HuguesDelorme, 2020-12-17:
I'm thinking of a more flexible solution by using a hierarchy of `XCAFPrs_AISObject`s instead a of a single one.
Those objects would form a graph matching closely the model tree. Instance shapes would be displayed with `AIS_ConnectedInteractive` objects.

SudatiSimone, 2020-12-17:
To understand If I realize the problem.
Now the changes that I made on the colour's component aren't permanent but If I close and re-open the file, there aren't my changes.  It's correct?

In my case, I want to open a step file, to do some changes (colour, name, attributes) and save the file with changes.
Maybe to solve the problem:
- open step file
- do changes
- file->export selected items (the root with stp extension) -> save as a new file with changes
Or also in this way, there is the same problem?

[image]

Now I can't do it because It gives me this error when I try to save it.

HuguesDelorme, 2020-12-17:
You get this error when trying the version in branch `feature/color` ?

SudatiSimone, 2020-12-17:
In **default branch**: it works but non properly.
On the left the starter file. I change a name and I export the file. On the right the exported file.
[image] [image]

In **feature/color branch** : it doesn't work and It gives me this error "Error during export to 'C:\....\name_file' File write problem".

HuguesDelorme, 2020-12-17:
It might be caused by your Export/STEP settings.
Run `Tools->Options` and try to "Restore Defaults" or at least make sure that `Schema` option is set to `AP214_IS` as below :
[image]

SudatiSimone, 2020-12-17:
Thanks again! 😍😍
With the trick of saving a new file with "Export selected items" I can create a new file with the changes I made. So the colour, name etc. that I choose is permanent!

In the next days, I will try to automatize it. Maybe with a simple "save" button that it does what "Export selected items" does. The save button will delete the old file and replace it with the new one

SudatiSimone, 2020-12-22:
I have the last question. 
I understand how now it works: I can colour for product and not for a single instance of the objects.
My new question is: 
it is possible to colour a single surface of objects?
For example to colour only one of the six faces of a cube.
What I want to understand is if it is feasible and with how much effort.
Then I'd work on it myself.

Thanks again

HuguesDelorme, 2020-12-27:
Yes this is possible, not yet directly with Mayo UI though.
You will need `XCAFDoc_ShapeTool` and `XCAFDoc_ColorTool` objects that you can access with `Mayo::Document::xcaf()` and `Mayo::XCaf::shapeTool/colorTool()`
Let's assume that the "cube" shape is stored in label `labelCube` in the document.
1. Declare the cube face `face` you want of a different color by a call to `XCAFDoc_ShapeTool::AddSubShape(labelCube, face)`. This will return the label `labelFace`
2. Call `XCAFDoc_ColorTool::SetColor(labelFace, color)`
3. Redisplay the 3D graphics

HuguesDelorme, 2021-01-05:
Hello Simone,

    Not related to our problem but I would like to ask you the resolution in pixels of your screen, because from the screenshots you posted in this thread I saw that the 3D viewcube and some icons are very small.
    I'm preparing a fix for that issue.

Thanks !

SudatiSimone, 2021-01-05:
Hi!
[image]
Looking at the technical specifications it says so but my feeling is that it's a lower resolution.
If you need some example's screens to see how It works on my pc I will send them to you.

closed 2021-01-25 by SudatiSimone