fougue/mayo #44: build with VS2019

smthan, 2020-12-13:
when i build the project with Opencascade7.5 + Qt5.12 + VS2019, there a 9 errors :
错误	LNK2019	无法解析的外部符号 "public: static class std::unique_ptr<class Mayo::PropertyGroup,struct std::default_delete<class Mayo::PropertyGroup> > __cdecl Mayo::IO::OccGltfReader::createProperties(class Mayo::PropertyGroup *)" (?createProperties@OccGltfReader@IO@Mayo@@SA?AV?$unique_ptr@VPropertyGroup@Mayo@@U?$default_delete@VPropertyGroup@Mayo@@@std@@@std@@PEAVPropertyGroup@3@@Z)，函数 "public: virtual class std::unique_ptr<class Mayo::PropertyGroup,struct std::default_delete<class Mayo::PropertyGroup> > __cdecl Mayo::IO::OccFactoryReader::createProperties(struct Mayo::IO::Format const &,class Mayo::PropertyGroup *)const " (?createProperties@OccFactoryReader@IO@Mayo@@UEBA?AV?$unique_ptr@VPropertyGroup@Mayo@@U?$default_delete@VPropertyGroup@Mayo@@@std@@@std@@AEBUFormat@23@PEAVPropertyGroup@3@@Z) 中引用了该符号	mayo	C:\Users\Administrator\Desktop\Qt\mayo-ci-occ7.5\io_occ.obj	1

please help me

HuguesDelorme, 2020-12-13:
That's weird, latest automatic build is fine ([appveyor](https://ci.appveyor.com/project/HuguesDelorme/mayo) CI both with OpenCascade 7.4 and 7.5)
I tried again with latest changes and build is also OK on my side.
You said "there are 9 errors" can you post here all of them ? I guess you just posted the last one.

smthan, 2020-12-14:
After add io_occ_gltf_reader.cpp & io_occ_gltf_writer.cpp into the vs project manually, it was builded OK.
But when start mayo.exe, whatever new or open, it was running death ... ...

smthan, 2020-12-14:
I have build "master" with opencascade7.3 , it work very well ... ...

HuguesDelorme, 2020-12-14:
The `master` branch is 5 months old against `develop` it would be better if you could use latest version !
Can you describe how do you build mayo ? How did you install OpenCascade 7.5 ?

smthan, 2020-12-14:
1. install opencascade 7.5
2. use Qt vs Tools to convert mayo.pro to mayo.sln
3. add io_occ_gltf_reader.cpp & io_occ_gltf_writer.cpp into the vs project
4. build is ok
5. run mayo.exe open a stp file , it always loading 0% ...

smthan, 2020-12-14:
It seems like the opencascade7.5 have some trouble ...
start example in mfc folder, like viewer3d.exe ,it running death
.....
opencascade7.4 or 7.3 viewer3d.exe running very well
so, how could i build mayo in opencascade 7.4 ?
--------------

smthan, 2020-12-14:
oh, my god!
build in opencascade7.4 successed! 
important : it work very well
mark：
exclude io_occ_gltf_write.cpp & io_occ_gltf_write.h from the vs project

closed 2020-12-14 by smthan