fougue/mayo #45: Open file Error : IFSelect_RetError

sinfee, 2020-12-28:
When I open this step file ,it got a error .
[jdzl-step.zip](https://github.com/fougue/mayo/files/5746208/jdzl-step.zip)

HuguesDelorme, 2020-12-28:
Mayo currently doesn't support ZIP files, you have to extract first the files from the archive.
[image]

sinfee, 2020-12-29:
I know but I havn't find a way to upload a step file. so I zip it. 
I find the error is because my file is under a chinese name directory  . 
[image]

HuguesDelorme, 2020-12-29:
Are you running a recent version of Mayo ? A couple of weeks ago I remember having adapted IO source code to use utf8 instead of local8bit for file paths.
Can you please check with latest version (in branch `develop`)
Let me know the result

HuguesDelorme, 2020-12-30:
Just did this test with version as of commit 2ec8230164a7268f96eb757bd4c3e5c440b179ee + OpenCascade 7.4.0 on Windows10 : open a STEP file located in the folder `F:\dev\data\test-encoding\简化的中文命名目录测试`
No problem reported :
[image]

sinfee, 2020-12-31:
Thanks for your help . I used release 0.2.0 . I will try current develop version.

closed 2021-01-03 by HuguesDelorme