fougue/mayo #48: Undefined reference to BSplCLib::MovePointAndTangent

VictorLamoine, 2021-01-14:
- Kubuntu 20.04
- OpenCascade 7.5.0 compiled from source
- Qt 5.12.8

```bash
export CSF_OCCTIncludePath=/home/victor/software/opencascade-7.5.0/install/include/opencascade/
export CSF_OCCTLibPath=/home/victor/software/opencascade-7.5.0/install/lib/
qmake ../src
git version 2.25.1
Project MESSAGE: Mayo version 0.3.0.605-cbee613 release
Project MESSAGE: Qt version 5.12.8
Project MESSAGE: OpenCascade version 7.5.0
Project MESSAGE: gmio OFF
make -j12
```

```bash
$ ./mayo 
./mayo: symbol lookup error: /lib/x86_64-linux-gnu/libTKG2d.so.7: undefined symbol: _ZN8BSplCLib19MovePointAndTangentEdRK8gp_Vec2dS2_dibiiRK18NCollection_Array1I8gp_Pnt2dERKS3_IdESA_RS5_Ri

$ c++filt _ZN8BSplCLib19MovePointAndTangentEdRK8gp_Vec2dS2_dibiiRK18NCollection_Array1I8gp_Pnt2dERKS3_IdESA_RS5_Ri
BSplCLib::MovePointAndTangent(double, gp_Vec2d const&, gp_Vec2d const&, double, int, bool, int, int, NCollection_Array1<gp_Pnt2d> const&, NCollection_Array1<double> const&, NCollection_Array1<double> const&, NCollection_Array1<gp_Pnt2d>&, int&)
```

HuguesDelorme, 2021-01-14:
`BSplCLib` is defined in `TKMath` package(TKMath.so)
Indeed the symbol lookup error indicates that the system doesn't search in the folder pointed to be `CSF_OCCTLibPath`
Does the path `/home/victor/software/opencascade-7.5.0/install/lib` appear in your `LD_LIBRARY_PATH` ? Shouldn't be required though as RPATH is used for that (see `opencascade.pri`)
Will try to build and run on a Linux PC and let you know

VictorLamoine, 2021-01-14:
I've got some progress (see https://github.com/fougue/mayo/compare/develop...VictorLamoine:develop)
```bash
/usr/bin/ld: application.o: undefined reference to symbol '_ZN34BinDrivers_DocumentRetrievalDriver16ReadShapeSectionER27BinLDrivers_DocumentSectionRSibRK21Message_ProgressRange'
/usr/bin/ld: /home/victor/software/opencascade-7.5.0/install/lib//libTKBin.so.7: error adding symbols: DSO missing from command line
```

It's now linking against the correct library but there is still something missing

VictorLamoine, 2021-01-14:
Any reason you use `qmake` instead of `CMake` ? I think the project would be easier to manage with `CMake`, and `qmake` will probably disappear in the future!

https://www.qt.io/blog/qt-6-build-system

> At the moment the qmake build system is still in place.

HuguesDelorme, 2021-01-14:
> Any reason you use `qmake` instead of `CMake` ? I think the project would be easier to manage with `CMake`, and `qmake` will probably disappear in the future!

Yes CMake is a better choice for the long-term. I started some weeks ago a port to CMake but this requires more work than expected, I have personally less know-how in CMake than qmake.
I always liked qmake. Honestly for developing Qt applications it will always feel simpler and more direct than CMake, it was tailored for that ...

HuguesDelorme, 2021-01-14:
> See [ce5ab9a](https://github.com/fougue/mayo/commit/ce5ab9a935f55a880249d74d05103e6f3c23b0a2) for my fix to link against the correct libraries. `LIBS += $$system_path($$join(CASCADE_LIB_DIR, " -L", -L))` does not seem to work and looks pretty hacky 😄

Could you try to add `message($$system_path($$join(CASCADE_LIB_DIR, " -L", -L)))` after this `LIBS += ...` line and check the output ?
It's working correctly with Windows builds and tried successfully in the past with GCC/Linux builds (some people could also build Mayo on macOS).
It looks "hacky" because of the syntax of qmake's `join()` function, here it is prepending the contents of variable `CASCADE_LIB_DIR` with `-L`
The surrounding `system_path()` function converts to native dir separators and by experience Windows/MSVC is touchy about this ...

Sorry for the inconvenience but I'm not sure if Mayo's qmake build script is the cause of your problems here.

VictorLamoine, 2021-02-12:
Hey there,

```bash
git version 2.25.1
Project MESSAGE: Mayo version 0.3.0.645-316bc06 release
Project MESSAGE: Qt version 5.12.8
Project MESSAGE: ---
Project MESSAGE: CASCADE_INC_DIR = /home/victor/software/opencascade-7.5.0/install/include/opencascade/
Project MESSAGE: CASCADE_LIB_DIR = /home/victor/software/opencascade-7.5.0/install/lib/
Project MESSAGE: CASCADE_BIN_DIR =
Project MESSAGE: CASCADE_SRC_DIR =
Project MESSAGE: ---
Project MESSAGE: OpenCascade version 7.5.0
Project MESSAGE: -L/home/victor/software/opencascade-7.5.0/install/lib/
Project MESSAGE: gmio OFF
```

Now I have a compile error because of the `libtree` template:
```bash
In file included from ../src/src/base/document_tree_node.h:10,
                 from ../src/src/base/document_tree_node_properties_provider.cpp:8:
../src/src/base/libtree.h: In function ‘void Mayo::traverseTree_unorder(const Mayo::Tree<T>&, const FN&)’:
../src/src/base/libtree.h:237:16: error: need ‘typename’ before ‘Mayo::Tree<T>::TreeNode’ because ‘Mayo::Tree<T>’ is a dependent scope
  237 |     for (const Tree<T>::TreeNode& node : tree.m_vecNode) {
      |                ^~~~~~~
      |                typename 

```

I also started to write a CMake script: https://github.com/VictorLamoine/mayo/blob/develop/CMakeLists.txt
Same compile error as the qmake project for the moment, I'm pretty sure the CMake script does not work but I'll work on it when i have some spare time!

HuguesDelorme, 2021-02-13:
Hello Victor,
Sorry for the compilation error (GCC?). Try what the compiler suggests to solve the issue :
```c++
     for (const typename Tree<T>::TreeNode& node : tree.m_vecNode) {
````
If it works then could you provide a PR ? I'll be pleased to merge it.

Your `CMakeLists.txt` looks promising, many thanks for the effort. I would suggest to use file globbing when specifying the headers/sources to be compiled, don't you think so ?

VictorLamoine, 2021-02-14:
`gcc 9.3.0`, pull request is [over there](https://github.com/fougue/mayo/pull/51), I should have read the error with more attention it was pretty straightforward!

Using glob to parse source files is discouraged with CMake, for now I'll keep it like it and we can discuss that when the CMake script works 😝

HuguesDelorme, 2021-03-10:
@VictorLamoine
How are you dealing with this build issue ? Did you solve it ?

VictorLamoine, 2021-03-11:
Sorry not yet, I did not take time to improve the CMake yet!

HuguesDelorme, 2021-06-03:
Can we close this issue as you could build Mayo with "official" OpenCascade packages(from https://github.com/fougue/mayo/issues/30#issuecomment-853346862) ?

VictorLamoine, 2021-06-03:
Yes let's close it

closed 2021-06-03 by VictorLamoine