fougue/mayo #53: Migrate to CMake

HuguesDelorme, 2021-02-23:
Current build system is QMake and it's still there for Qt6.
In the long-term the Mayo project should migrate to CMake

luzpaz, 2022-09-09:
Any ETA on this?

HuguesDelorme, 2023-03-29:
Started to work on that task(see branch [port/cmake](https://github.com/fougue/mayo/tree/port/cmake))
But has to halt because the `cmake/OpenCASCADEConfig.cmake` file provided with OpenCascade is broken (for all versions starting from 7.3.0)
This was reported in [OpenCascade forums](https://dev.opencascade.org/content/3rdparty-hard-coded-paths-when-building-against-opencascade)

HuguesDelorme, 2024-02-29:
See updated build instructions for CMake:
[Windows](https://github.com/fougue/mayo/wiki/Build-instructions-for-Windows)  
[Linux](https://github.com/fougue/mayo/wiki/Build-instructions-for-Linux)  
[macOS](https://github.com/fougue/mayo/wiki/Build-instructions-for-macOS)

closed 2024-02-29 by HuguesDelorme