fougue/mayo #54: Correct support of High DPI monitors

HuguesDelorme, 2021-02-23:
Some items(ViewCube, buttons in the 3D view, ...) look small on High DPI(eg 4k) monitors.
This is caused by usage of hard-coded pixel sizes in the source code.

Find a way to express sizes that are compatible for any screen resolution.

prokoudine, 2023-01-27:
The same applies to measures display in the viewport (the panel is fine).

HuguesDelorme, 2023-02-10:
@prokoudine 
> The same applies to measures display in the viewport (the panel is fine).

Branch bugfix/highdpi is an attempt to fix high DPI support.
I don't have a suitable screen for real testing, I used QT_SCALE_FACTOR to simulate higher DPI.
It would be great if you have some time and interest to check that branch solves the issue on your local machine!
Let me know any feedback(eg screenshots)
Greetings

prokoudine, 2023-02-10:
The branch doesn't build here. All sorts of warnings, and it stops at this:

```
  128 | class PropertyGroupSignals : public PropertyGroup {
      |       ^~~~~~~~~~~~~~~~~~~~
In file included from ../mayo/src/app/mainwindow.h:11,
                 from ../mayo/src/app/app_context.cpp:9:
../mayo/src/app/../base/property.h:128:7: note: previous definition of ‘class Mayo::PropertyGroupSignals’
  128 | class PropertyGroupSignals : public PropertyGroup {
      |       ^~~~~~~~~~~~~~~~~~~~
In file included from src/app/widget_model_tree.h:10,
                 from ./ui_mainwindow.h:32,
                 from ../mayo/src/app/app_context.cpp:10:
src/app/../base/property.h:142:27: error: redefinition of ‘static bool Mayo::Property::setValueHelper(Mayo::Property*, T*, const T&)’
  142 | template<typename T> bool Property::setValueHelper(Property* prop, T* ptrValue, const T& newValue)
      |                           ^~~~~~~~
In file included from ../mayo/src/app/mainwindow.h:11,
                 from ../mayo/src/app/app_context.cpp:9:
../mayo/src/app/../base/property.h:142:27: note: ‘static bool Mayo::Property::setValueHelper(Mayo::Property*, T*, const T&)’ previously declared here
  142 | template<typename T> bool Property::setValueHelper(Property* prop, T* ptrValue, const T& newValue)
      |                           ^~~~~~~~
make: *** [Makefile:3059: app_context.o] Error 1
make: *** Waiting for unfinished jobs....

```

HuguesDelorme, 2023-02-13:
@prokoudine

What compiler are you using ? GCC9 ?
That's weird you get some errors as the ubuntu/macOS CI are building fine

prokoudine, 2023-02-13:
gcc (Ubuntu 11.3.0-1ubuntu1~22.04) 11.3.0

HuguesDelorme, 2023-02-14:
@prokoudine 
I can't find a reason why the build is failing on your side
The ubuntu CI is also using gcc 11.3
[image]

Is the `develop` branch building fine with the same environment?

HuguesDelorme, 2023-03-02:
Tested by @VictorLamoine

closed 2023-03-02 by HuguesDelorme