fougue/mayo #59: compile error

tyc1998, 2021-03-09:
I just followed the method on https://github.com/fougue/mayo to build mayo. However, many LINK2001 errors appeared during the generation.
Just like this: the unresolved external symbol "public: void __cdecl gp_Pnt::DumpJson(class std::basic_ostream<char,struct std::char_traits<char>> &,int)const "(?DumpJson@gp_Pnt@@QEBAXAEAV ?$basic_ostream@DU?$char_traits@D@std@@@std@@H@Z) mayo D:\cpp\stp\mayo\build\mainwindow.obj 1
And the errors are all caused by the obj file.
The opencascade version I use is 7.5.0, and the qt version is 5.15.0. I really don’t know what method can be used to solve this kind of error. I look forward to your help!

HuguesDelorme, 2021-03-09:
Hello @tyc1998
Are you using the OpenCascade pre-built DLLs (provided by the [Download Center](https://old.opencascade.com/content/latest-release)) or did you build them yourself ?
Are you just having linking problems ?
What version of Visual Studio are you using ?

tyc1998, 2021-03-09:
I downloaded opencascade in the download center. Since I don't know that there will be a pre-built dll, I opened OCCT.sln in D:\OpenCASCADE-7.5.0-vc14-64\opencascade-7.5.0\adm\msvc\vc141 to build. I don't know if it had an impact.
I only have LNK2001 and LNK2019 link problems.
The version of vs I use is visual studio 2017

HuguesDelorme, 2021-03-09:
I would delete your whole current OpenCascade 7.5.0 installation and run the setup again.
Then build Mayo.

tyc1998, 2021-03-09:
Thanks, I will try it.

HuguesDelorme, 2021-03-09:
Let me know of any other problem.
Notice you can build Mayo against the pre-built OpenCascade 7.5 DLLs the appveyor CI is using(see [here](http://www.fougue.pro/share/bin/OpenCASCADE-7.5.0-vc14-64.rar)). They are the same as from the OpenCascade Download Center but cleaned from unused materials.

Also you can try directly Mayo with the v0.3.0 installer, please see the [Releases](https://github.com/fougue/mayo/releases) section.
You won't have access to the [latest additions](https://github.com/fougue/mayo/milestone/4?closed=1) of current v0.4.0-dev though.

tyc1998, 2021-03-09:
There are still many link errors!
For example, the unresolved external symbol "private: static class NCollection_Vec3<float> __cdecl Quantity_Color::valuesOf(enum Quantity_NameOfColor,enum Quantity_TypeOfColor)" (?valuesOf@Quantity_Color@@CA?AV?$NCollection_Vec3@M@@W4TypeColor@W4 @@@Z) mayo D:\cpp\stp\mayo\build\widget_clip_planes.obj

I have reinstalled opencascade, and I tried both Qt5.15.0 and Qt5.13.0, and the problem exists.

Below is a screenshot of my command line
[image]
[image]

I have a headache, I don’t know where the problem is, I look forward to your help

tyc1998, 2021-03-09:
During the build process, visual studio prompts 1>..\src\app\widget_shape_selector.h(0): Note: No relevant classes found. No output generated.
I don't know if the problem is here

tyc1998, 2021-03-09:
Thanks for your help!
Compared with ready-made executable programs, my purpose is more inclined to learn your code. Hope you can solve this problem

tyc1998, 2021-03-09:
By the way, does my situation seem to find the opencascade libs?

grotius-cnc, 2021-03-09:
@Tyc,

in your build directory as showed in picture above.

For me it's like this : 

$ mkdir build
$ cd build
$ /usr/local/include/qt/5.15.1/gcc_64/bin/qmake /home/user/mayo/mayo.pro

To explain above line. 
	First argument is the path to the qmake executable of qt-creator-c++
	Second argument is the path to the mayo.pro file. (qt project file)
	
	Terminal output  : 
			git version 2.20.1
			Project MESSAGE: Mayo version 0.3.0.491-8f89325 release
			Project MESSAGE: Qt version 5.15.1
			Project MESSAGE: gmio OFF
			Project MESSAGE: OpenCascade version

	This will output a Makfile in the build directory, then we can build the App with :
	
$ make

grotius-cnc, 2021-03-09:
@tyc,

_my purpose is more inclined to learn your code_
Do you mean to learn opencascade? or in generai the way this program is written in QT?

tyc1998, 2021-03-09:
My purpose is to use qt and opencascade to write a small tool that can read stp file information and realize the selection of point and surface information.

grotius-cnc, 2021-03-09:
Ahh. Then you have to look at a more simple project, to understand how opencascade is used by more simple examples.
Take a look at this : https://github.com/grotius-cnc/Skynet_Robot_Control_Rtos_Ethercat/releases/tag/1.0.18

If you delete the hal & kineatics stuff, you got a very nice starting point. It's used by more people to start with.
One comment of a user : **Thank you for this cool project. It really helps to learn how to use opencascade and robot kinematic.**

grotius-cnc, 2021-03-09:
@Tyc,

Realize yourself the Mayo qt c++ code is not for beginner programmers. I can say it's not my style, with respect to to the writer.
Reverse engeneering the mayo code will take you day's. And you will end up with something that i provided by the above link,
where all opencascade function's are clear and understandable written down in c++.

HuguesDelorme, 2021-03-09:
@tyc1998
From your qmake output, it seems like there's a problem with finding OpenCascade (see the output line "`Project MESSAGE: OpenCascade version`" the version info isn't shown)
Please could you post here the output of the command `set CSF_` so we can see all OpenCascade paths and check there are fine.

Also make sure to start Visual Studio with the OpenCascade and Qt environments correctly set

HuguesDelorme, 2021-03-09:
@grotius-cnc
Thanks for reporting the compilation error, actually this is related to a recent change that isn't compatible with OpenCascade v7.3 (the functions `Quantity_Color::ColorToHex()` and `ColorFromHex()` appeared in v7.4)
I will provide a fix and let you know when it's done.

HuguesDelorme, 2021-03-09:
@grotius-cnc
Fixes commited to develop branch, let me know if you have other compilation problems with Linux/GCC

grotius-cnc, 2021-03-10:
Hi,  

Just did a fresh clone and compiled. 
Only added my includepath to the pro file.
 _/usr/local/include/opencascade/_

Attached a picture of compile errors.
It looks the most are related to a magic_enum.

Is this error not been noticed when compiling on windows?

[image]

HuguesDelorme, 2021-03-10:
@grotius-cnc 
Mayo is internally using [magic_enum](https://github.com/Neargye/magic_enum)
And this library requires GCC >= 9
Please check you GCC version with `gcc --version`

grotius-cnc, 2021-03-15:
Hi,
Thanks, that one is solved. But then i get second error.
Currently building a new debian 11 distro. When this is finished i will try to get Mayo compiled on this new distro.
This week Mayo has 31 downloads for linux.

grotius-cnc, 2021-04-24:
Hi,

Today i compiled mayo again with opencascade 7..5.0 beta on Linux-Pro 

It looks the compile output is not showing the opencascade version number.
An idea: can you add a line in the .pro file where users can add this info manually?

I am not sure, but i think the opencascade library path name is not fixed. Is it alway's installed under the name  
"/usr/local/include/opencascade/" ? If so for linux, then this path could be fixed in the .pro file. 

In the io_occ.cpp file i had to comment out all the "#if OCC_VERSION_HEX >= OCC_VERSION_CHECK(7, 5, 0)" code chucks at 7 places. Somehow it's not accepting the 7.5.0

In the .pro file i added a few common linux includepaths, maybe add something like "if unix .. then include .. " :
INCLUDEPATH +=  /usr/local/include/opencascade/ \
                /usr/include/ \
                /usr/local/lib/ \

That's all. Succesfully compiled !

[image]

When i start Mayo i get : libTKBin.so.7: cannot open shared object file
I made a postinst script where it creates the /etc/ld.so.conf.d/ lib path's and then doing a ldconfig.

Here is the deb archive :100: 
https://github.com/grotius-cnc/Linux-Pro/releases/download/1.0.0/mayo-cad-viewer.deb
install : sudo dpkg -i mayo-cad-viewer.deb
Will install in the /opt/ folder. Will create a desktop launcher. 
The source code and qt project is included in the .deb package as reference material.

HuguesDelorme, 2021-04-26:
@grotius-cnc 
Path to OpenCascade include and lib dirs aren't hardcoded in mayo.pro
Have a look at opencascade.pri : 
- OpenCascade environment variables(CSF_OCCTIncludePath, CSF_OCCTLibPath, ...) are used if defined(usually with env.bat on Windows)
- You can force OpenCascade paths with QMake variables CASCADE_INC_DIR and CASCADE_LIB_DIR, ie set those variables when calling qmake from the command line(see example in https://github.com/fougue/mayo/blob/develop/scripts/travis-build.sh)

In your case you can try :
`qmake CASCADE_INC_DIR=/usr/local/include/opencascade ../mayo.pro`

About the error libTKBin.so.7 maybe this is related with how you built OpenCascade ?
Linux Travis build is using "official" OpenCascade 7.3 packages

HuguesDelorme, 2022-03-22:
No activity

closed 2022-03-22 by HuguesDelorme