fougue/mayo #60: Compile error 'call to 'test' is ambiguous'

k0d, 2021-03-21:
I'm trying to compile on MacOS 11 (BigSur), I've fixed most of the issues but I've come across one that I don't understand.

The code here
https://github.com/fougue/mayo/blob/develop/src/io_occ/io_occ.cpp#L78
is calling the code here
https://github.com/fougue/mayo/blob/develop/src/io_occ/io_occ.cpp#L98
which fails with the error message below.

I'm not 100% sure about templates and constexpr, so I'm stuck on how to progress with this.

(Nice work by the way...this is the app I needed in my life :) )

`src/io_occ/io_occ.cpp:106:46: error: call to 'test' is ambiguous
        static constexpr bool value = sizeof(test<Derived>(nullptr)) == sizeof(Yes);
                                             ^~~~~~~~~~~~~
src/io_occ/io_occ.cpp:78:53: note: in instantiation of static data member 'Mayo::IO::(anonymous
      namespace)::FactoryData<Mayo::IO::Writer>::HasMember_createProperties<Mayo::IO::OccBRepWriter>::value' requested here
        if constexpr(HasMember_createProperties<T>::value) {
                                                    ^
src/io_occ/io_occ.cpp:139:10: note: in instantiation of function template specialization 'Mayo::IO::(anonymous
      namespace)::FactoryData<Mayo::IO::Writer>::addExchanger<Mayo::IO::OccBRepWriter>' requested here
        .addExchanger<OccBRepWriter>(Format_OCCBREP)
         ^
src/io_occ/io_occ.cpp:103:41: note: candidate function [with U = Mayo::IO::(anonymous
      namespace)::FactoryData<Mayo::IO::Writer>::HasMember_createProperties<Mayo::IO::OccBRepWriter>::Derived]
        template<typename U> static No& test(decltype(U::createProperties)*);
                                        ^
src/io_occ/io_occ.cpp:104:42: note: candidate function [with U = Mayo::IO::(anonymous
      namespace)::FactoryData<Mayo::IO::Writer>::HasMember_createProperties<Mayo::IO::OccBRepWriter>::Derived]
        template<typename U> static Yes& test(U*);
                                         ^
src/io_occ/io_occ.cpp:78:22: error: constexpr if condition is not a constant expression
        if constexpr(HasMember_createProperties<T>::value) {
                     ^
src/io_occ/io_occ.cpp:139:10: note: in instantiation of function template specialization 'Mayo::IO::(anonymous
      namespace)::FactoryData<Mayo::IO::Writer>::addExchanger<Mayo::IO::OccBRepWriter>' requested here
        .addExchanger<OccBRepWriter>(Format_OCCBREP)`

HuguesDelorme, 2021-03-22:
@k0d 
Thanks for reporting this Clang build error, and apologizes for this!
I knew that template machinery in `io_occ.cpp` would be fragile regarding compiler portability.
I've just pushed a fix in branch `buildfix/#60` could you please get it and try if build suceeds on your PC ?
Feel free to report any further build problem here.

k0d, 2021-03-30:
Thanks for the fix, it works fine.

closed 2021-03-30 by k0d