fougue/mayo #64: error during import of  step file transfer problem

arthurx70, 2021-04-25:
win7 system setup with vs2019+qt5.14.2 and occt7.5.0,no problem in compile，and no problem with iges brep importing，but step importing with error "error during import of.....transsfer problem"

HuguesDelorme, 2021-04-26:
Hello @arthurx70 
Can you provide here the STEP file that causes the error ?

arthurx70, 2021-04-26:
I sure its not the problem of step file.I have test the example step file in occt source folder data/step*.I have test the same step file on   win10，it work without error. I think maybe occt47.5.0was not support very well.by the way i find the std::filesystem was not work well when the file path was not in utf8....

arthurx70, 2021-04-26:
importing step file，I find after XSControl_reader readfile，then transfer shape，transfer cannot find binder!

arthurx70, 2021-04-26:
I have test on another win7 system，the same error appear，I think it may is a bug，someone needed to fix it.my environment is win7pro，vs2019+qt5.14.2，I commpiled occt7.5.0

HuguesDelorme, 2021-04-26:
@arthurx70 
I'm not sure to understand the problem, can you give more details ?
What's the name of the STEP file (you mentionned utf8-encode filepath that might cause problems) ?

arthurx70, 2021-04-27:
first，the error during importing step file transfer problem，is because the os is win7，if you have a computer with win7 os，and install vs2019+qt5.14.2，compile the occt7.5.0，compile mayo and import any step file will appear this error.the step file u can use any one you you have，also you can use the step files in occt source file folder，like screw...step.second problem，if I have import a file with path is invalid，it cause a failure broken at the filesystem.and every time i start up it broken down，can not into the mayo start page.

arthurx70, 2021-05-01:
this bug can be fixed？

HuguesDelorme, 2021-05-01:
@arthurx70 

For the first problem, do you have also that issue with the v0.3 installer (see [Releases page](https://github.com/fougue/mayo/releases)) ?

The second problem you reported should be fixed with https://github.com/fougue/mayo/issues/63
As you build Mayo yourself, then make sure you have latest changes of `develop` branch.
If that problem persists then let me know.
As a temporary solution you can clear value `HKEY_CURRENT_USER\SOFTWARE\Fougue Ltd\Mayo\application\recentFiles` in registry

arthurx70, 2021-05-02:
unfortunately!two problems are not fixed.
first problem,I have test the v0.3 installer (see Releases page) ,also the v0.3 source code, the problem is exist all the way,this problem is only occur on win7 os,on win10 it works well.
second problem，If I have load file with the path encode like japanese or Chinese，it crashed,and I try to start it again,it crashed all the time,this problem is occur on both win10 and win7.

arthurx70, 2021-05-02:
I have test a older version have no the first problem,which not use FilePath(std::filesystem::path),just use QString for file path.

arthurx70, 2021-05-02:
I have tested an older version which have no first problem,this version used QString for file path,but not FilePath(std::filesystem::path);

HuguesDelorme, 2021-05-03:
@arthurx70 
I've reproduced the bug you reported affecting `FilePath` and fixed it in `develop` branch(see commit 3717ef14de682ecc2f145057750d3099d3e19031)
Let me know if it works as well on your side

I will try to reproduce the Win7 issue during this week.

arthurx70, 2021-05-05:
@HuguesDelorme ，the path  ecode problem is fixed，I test it works well with Chinese/japanese path,but the importing step files with transfer problem also exist.

arthurx70, 2021-05-11:
@HuguesDelorme ,do u have reproduce the step file transfer problem on win7 
on ur side?

arthurx70, 2021-05-11:
did you have compiled on win7,I guess you just copy the exe into win7，maybe.

HuguesDelorme, 2021-05-11:
Yes I've built Mayo on Win10, generated a setup and then installed on Win7(running from VM)
At first glance I don't see why building on Win7 would make a difference ?
Same compiler, same OpenCascade&Qt binaries

I suggest you run Mayo with debug OpenCascade binaries and step into the Transfer() call to figure out what's wrong.

diplodocuslongus, 2021-05-28:
I have the same `Error during import [...] File transfer problem` message when trying to load a STEP file (I've tried various step files, the cube.step in the `/test/inputs` folder of mayo does this as well), the iges and brep files I've tried can be imported. 
My system:
- Ubuntu 18.04
- gcc 9.3
- QT 5.15
- OpenCascade version 7.5 built from source
I've tried both mayo v0.3.0 and the develop branch (updated today), both built from source.

diplodocuslongus, 2021-05-31:
@HuguesDelorme This solved it! At least on my platform/configuration (ubuntu 18.04, mayo develop branch you mentioned, opencascade 7.5 built locally and QT 5.15). I can now import step files without any issue. Thanks!

HuguesDelorme, 2021-05-31:
@diplodocuslongus 
Good news! I will wait this week for @arthurx70 feedback to close this issue

closed 2021-06-07 by HuguesDelorme