fougue/mayo #66: "File transfer problem" Error when debug mode,but release has no problem

dansirming, 2021-05-04:
dear expert：
     recently I encounter a  problem! that is : some step files are opened by mayo under debug mode,pop up "File transfer problem"
error prompt, but has no problem under release mode, I has positioned "src/io_occ/io_occ_caf.cpp/cafGenericReadTransfer/reader.Transfer",
I am so sure that the step file is correct!  the network is so bad, so step file can not upload ! can you help me ? thank you!

[888.zip](https://github.com/fougue/mayo/files/6419940/888.zip)

dansirming, 2021-05-04:
888.zip is my step file ,you can use it test,thank you !

dansirming, 2021-05-04:
my environment ：Qt  5.14.1,  OpenCASCADE-7.5.0   Or    OpenCASCADE-7.3.0

HuguesDelorme, 2021-05-04:
Hello @dansirming 
Are you using latest version of `develop` branch ?
I think the issue you reported is fixed with #65

HuguesDelorme, 2021-05-04:
Loaded `888.STEP` with Mayo built from `develop` branch :
[image]

dansirming, 2021-05-04:
I download "mayo-717ef14de682ecc2f145057750d3099d3e19031",but still pop up "File transfer problem", can you tell me you use which version ? thank you very much !

dansirming, 2021-05-04:
I download "mayo-717ef14de682ecc2f145057750d3099d3e19031",but still pop up "File transfer problem" under debug mode, can you tell me you use which version ? thank you very mu
[image]
ch !

dansirming, 2021-05-04:
I download "mayo-3717ef14de682ecc2f145057750d3099d3e19031",but still pop up "File transfer problem" under debug mode,  but has no problem under release mode, can you tell me you use which version ? thank you very much!
[image]
ch !

HuguesDelorme, 2021-05-04:
That seems latest version from the commit id you give.
Seems like you build Mayo with OpenCascade 7.3 ?
Do you have the same issue with OpenCascade 7.5 ? This is the version I'm using(from OpenCascade's Download center)

dansirming, 2021-05-04:
I Has tried many times!  both OpenCascade 7.3  and OpenCascade 7.5 all pop up  "File transfer problem" under debug mode

dansirming, 2021-05-04:
> from OpenCascade's Download center

I download from OpenCascade's Download center and will try soon! thank you very much

HuguesDelorme, 2021-05-04:
OpenCascade Download Center ;
https://dev.opencascade.org/release

dansirming, 2021-05-04:
I Has tried just now, from OpenCascade Download Center https://dev.opencascade.org/release,pop up "File transfer problem" under debug mode 

[image]

dansirming, 2021-05-04:
reader.Transfer return flase undet debug mode!
[image]

dansirming, 2021-05-04:
I Has tried just now, from OpenCascade Download Center https://dev.opencascade.org/release,pop up "File transfer problem" under debug mode ,why has no problem under release mode?

 

 https://dev.opencascade.org/release

dansirming, 2021-05-04:
reader.Transfer return flase under debug mode!

 

 https://dev.opencascade.org/release

dansirming, 2021-05-04:
thank you for help me! Wish you all happiness, health and prosperity！！！

 

 https://dev.opencascade.org/release

HuguesDelorme, 2021-05-04:
Problem still exist with "official" OpenCascade binaries ?
What happens if you try with Mayo v0.3 installer(see Releases [page](https://github.com/fougue/mayo/releases/tag/v0.3.0)) ?
Sorry that's not easy to help, as I didn't reproduce your issue on my side.

dansirming, 2021-05-04:
with "official" OpenCascade binaries has no problem ，because binaries is compiled out under release mode by source code, but  with "official" OpenCascade "Mayo"  source code(as below picture) compiled out a binaries that  under debug mode ,the  issue  will reproduce ! you can download  "official" OpenCascade Mayo source code(as below picture)  and try to use 888.zip step file under debug mode ,then you can reproduce this issue，bless you good luck！

HuguesDelorme, 2021-05-04:
I tried with OpenCascade 7.5.0 built from source code in debug configuration:
3RDPARTY_DIR=<rootdir_of_official_occ7.5>
BUILD_USE_PCH=ON
BUILD_WITH_DEBUG=ON
BUILD_LIBRARY_TYPE=Shared
BUILD_YACCLEX=OFF
USE_FFMPEG=OFF
USE_FREEIMAGE=ON
USE_GLES2=OFF
USE_OPENVR=OFF
USE_RAPIDJSON=ON
USE_TBB=ON
USE_VTK=OFF
...

I can open file `888.STEP` without problems.
Can you step into the call `reader.Transfer()` that return false ? You should see what's wrong there

dansirming, 2021-05-05:
“reader.Transfer()”  in the lib， I can't step into it！it just return false。 
I do mayo project by Visual studio 2017 use Qt Vs Toos plug-in(Qt core is 5.14.1), 
now I send my mayo project  to you,can you help me ? may be you can reproduce my issue on your side.thank you very much

 

 3RDPARTY_DIR=<rootdir_of_official_occ7.5&gt;

dansirming, 2021-05-10:
dear expert:
     today , I  extract code from mayo project into myself project to satisfied my need , the  problems which I encountered last week has not appeared， I guess that some configs Which I maked are not correct before . thank you very much!

arthurx70, 2021-05-11:
@dansirming,hello,can U tell me how to fix the problem, importing step file occur with transfer problem. did U compiled on win7 os? what configs U have changed?

dansirming, 2021-05-16:
dear expert:
     last week so busy that I has no time to send mail to you. so sorry ！
    last week  I  extracted code from mayo project into myself project to satisfied my need , the  problems which I encountered last week has not appeared。the max changes as follow：

//    -----------------------io_occ_step.cpp--------------------------
bool OccStepReader::readFile(const FilePath&amp; filepath, TaskProgress* progress)
{
    MayoIO_CafGlobalScopedLock(cafLock);

	// 以下两句代码不要，否则debug模式下，有的文件打开会报错，release版则不会
    OccStaticVariablesRollback rollback;
    //this-&gt;changeStaticVariables(&amp;rollback);
    return Private::cafReadFile(m_reader, filepath, progress);
}

//    -----------------------io_occ_step.cpp--------------------------
bool OccStepReader::transfer(DocumentPtr doc, TaskProgress* progress)
{
    MayoIO_CafGlobalScopedLock(cafLock);

	// 以下两句代码不要，否则debug模式下，有的文件打开会报错，release版则不会
   OccStaticVariablesRollback rollback;
   // this-&gt;changeStaticVariables(&amp;rollback);
    return Private::cafTransfer(m_reader, doc, progress);
}

// --------------- my main widget class，   super3DShowWnd.cpp
	super3DShowWnd::super3DShowWnd(GuiApplication* pGuiApp, QWidget *parent)
		: QWidget(parent), m_guiApp(pGuiApp)
	{
		ui.setupUi(this);

		this-&gt;setWindowFlags(Qt::CustomizeWindowHint | Qt::FramelessWindowHint);
		setAttribute(Qt::WA_AcceptTouchEvents);

		app-&gt;settings()-&gt;resetAll();
		app-&gt;settings()-&gt;load();
		
	}

I canceled the red codes in the “io_occ_step.cpp” and  remove  "settings()-&gt;load()"  to the my main widget class ,  the  problems  has not appeared ! I tried to do that in mayo project, the  problems  has  still  appeared!
my environment as follow:

vs 2017、Qt 5.14.1、qt-vsaddin-msvc2017-2.4.1-rev.12.vsix，OpenCASCADE-7.5.0-vc14-64   Or  OpenCASCADE-7.3.0-vc14-64

HuguesDelorme, 2021-05-17:
@dansirming
One easy thing that's worth trying : edit settings in Mayo (menu Tools->Options) and restore default values for the "Import/STEP" section
[image]

dansirming, 2021-05-21:
I tried, but all the same!!!,the problem still  appear！

HuguesDelorme, 2021-05-28:
I've provided in `develop` branch a related fix(commit 5f71dd6)
Let me know if it fixes your problem.
Make sure, after compilation, to reset `Import/STEP` options to default values on first run(Tools->Options)

dansirming, 2021-06-06:
today， I tried，the last question has fixed! thank you very much!  but when I use opencascade-7.5.0 to bulid mayo,
some link errors as follows：

I has added all libraries which that  are located  “opencascade-7.5.0\win64\vc14\lib”   to mayo project，but all the same!  I remove the code  of "if OCC_VERSION_HEX &gt;= OCC_VERSION_CHECK(7, 4, 0)" in "io_occ.cpp", the question  disappear and never show up， can you help me to fix the question? thank you very much!!!

 

 Make sure, after compilation, to reset Import/STEP options to default values on first run(Tools-&gt;Options)

HuguesDelorme, 2021-06-06:
@dansirming 
I can't see the link errors in your last comment.

dansirming, 2021-06-07:
the link errors show in the pictures, can you help me ? thanks!

HuguesDelorme, 2021-06-07:
@dansirming
Link errors in the pictures ? But where are those pictures ?

dansirming, 2021-06-07:
pictures that in the Email attachment！but I don't know why you can't see them! now I past link errrors as follow:

: 无法解析的外部符号 "public: __cdecl Mayo::IO::OccGltfReader::OccGltfReader(void)" (??0OccGltfReader@IO@Mayo@@QEAA@XZ)，该符号在函数 "class std::unique_ptr<class Mayo::IO::OccGltfReader,struct std::default_delete<class Mayo::IO::OccGltfReader&gt; &gt; __cdecl std::make_unique<class Mayo::IO::OccGltfReader,0&gt;(void)" (??$make_unique@VOccGltfReader@IO@Mayo@@$$V$0A@@std@@YA?AV?$unique_ptr@VOccGltfReader@IO@Mayo@@U?$default_delete@VOccGltfReader@IO@Mayo@@@std@@@0@XZ) 中被引用
1&gt;io_occ.obj : error LNK2019: 无法解析的外部符号 "public: static class std::unique_ptr<class Mayo::PropertyGroup,struct std::default_delete<class Mayo::PropertyGroup&gt; &gt; __cdecl Mayo::IO::OccGltfReader::createProperties(class Mayo::PropertyGroup *)" (?createProperties@OccGltfReader@IO@Mayo@@SA?AV?$unique_ptr@VPropertyGroup@Mayo@@U?$default_delete@VPropertyGroup@Mayo@@@std@@@std@@***@***.***@@Z)，该符号在函数 "public: virtual class std::unique_ptr<class Mayo::PropertyGroup,struct std::default_delete<class Mayo::PropertyGroup&gt; &gt; __cdecl Mayo::IO::OccFactoryReader::createProperties(struct Mayo::IO::Format const &amp;,class Mayo::PropertyGroup *)const " (?createProperties@OccFactoryReader@IO@Mayo@@UEBA?AV?$unique_ptr@VPropertyGroup@Mayo@@U?$default_delete@VPropertyGroup@Mayo@@@std@@@std@@***@***.***@***@***.***@@Z) 中被引用
1&gt;io_occ.obj : error LNK2019: 无法解析的外部符号 "public: __cdecl Mayo::IO::OccObjReader::OccObjReader(void)" (??0OccObjReader@IO@Mayo@@QEAA@XZ)，该符号在函数 "class std::unique_ptr<class Mayo::IO::OccObjReader,struct std::default_delete<class Mayo::IO::OccObjReader&gt; &gt; __cdecl std::make_unique<class Mayo::IO::OccObjReader,0&gt;(void)" (??$make_unique@VOccObjReader@IO@Mayo@@$$V$0A@@std@@YA?AV?$unique_ptr@VOccObjReader@IO@Mayo@@U?$default_delete@VOccObjReader@IO@Mayo@@@std@@@0@XZ) 中被引用
1&gt;io_occ.obj : error LNK2019: 无法解析的外部符号 "public: static class std::unique_ptr<class Mayo::PropertyGroup,struct std::default_delete<class Mayo::PropertyGroup&gt; &gt; __cdecl Mayo::IO::OccObjReader::createProperties(class Mayo::PropertyGroup *)" (?createProperties@OccObjReader@IO@Mayo@@SA?AV?$unique_ptr@VPropertyGroup@Mayo@@U?$default_delete@VPropertyGroup@Mayo@@@std@@@std@@***@***.***@@Z)，该符号在函数 "public: virtual class std::unique_ptr<class Mayo::PropertyGroup,struct std::default_delete<class Mayo::PropertyGroup&gt; &gt; __cdecl Mayo::IO::OccFactoryReader::createProperties(struct Mayo::IO::Format const &amp;,class Mayo::PropertyGroup *)const " (?createProperties@OccFactoryReader@IO@Mayo@@UEBA?AV?$unique_ptr@VPropertyGroup@Mayo@@U?$default_delete@VPropertyGroup@Mayo@@@std@@@std@@***@***.***@***@***.***@@Z) 中被引用
1&gt;io_occ.obj : error LNK2001: 无法解析的外部符号 "public: virtual bool __cdecl Mayo::IO::OccGltfWriter::transfer(class gsl::span<class Mayo::ApplicationItem const ,-1&gt;,class Mayo::TaskProgress *)" (?transfer@OccGltfWriter@IO@Mayo@@UEAA_NV?$span@$$CBVApplicationItem@Mayo@@***@***.***@@***@***.***@@Z)
1&gt;io_occ.obj : error LNK2001: 无法解析的外部符号 "public: virtual bool __cdecl Mayo::IO::OccGltfWriter::writeFile(class std::filesystem::path const &amp;,class Mayo::TaskProgress *)" (?writeFile@OccGltfWriter@IO@Mayo@@***@***.***@std@@***@***.***@@Z)
1&gt;io_occ.obj : error LNK2001: 无法解析的外部符号 "public: virtual void __cdecl Mayo::IO::OccGltfWriter::applyProperties(class Mayo::PropertyGroup const *)" (?applyProperties@OccGltfWriter@IO@Mayo@@***@***.***@@Z)

HuguesDelorme, 2021-06-08:
Make sure your git working copy is clean(no pending change) and try to rebuild again ?
The Windows/MSVC appveyor CI is fine so you shouldn't have any problem with building Mayo develop branch

dansirming, 2021-06-12:
I have clean many times,but the link error all the same! what is "The Windows/MSVC appveyor CI"? I I've never heard of it， can you help me to install it , thank you very much!   the link error that is in the "io_occ.cpp" ,  it seem that  can not find the libraries 
about OccObjReader or about  OccGltfReader

std::unique_ptr<Reader&gt; OccFactoryReader::create(const Format&amp; format) const
{
    if (format == Format_STEP)
        return std::make_unique<OccStepReader&gt;();
    if (format == Format_IGES)
        return std::make_unique<OccIgesReader&gt;();
    if (format == Format_OCCBREP)
        return std::make_unique<OccBRepReader&gt;();
    if (format == Format_STL)
        return std::make_unique<OccStlReader&gt;();

// the  following code will produce some link erros
#if OCC_VERSION_HEX &gt;= OCC_VERSION_CHECK(7, 4, 0)
    if (format == Format_GLTF)
        return std::make_unique<OccGltfReader&gt;();
    if (format == Format_OBJ)
        return std::make_unique<OccObjReader&gt;();
#endif

    return {};
} 

PtrPropertyGroup OccFactoryReader::createProperties(const Format&amp; format, PropertyGroup* parentGroup) const
{
    if (format == Format_STEP)
        return OccStepReader::createProperties(parentGroup);
    if (format == Format_IGES)
        return OccIgesReader::createProperties(parentGroup);

// the  following code will produce some link erros
#if OCC_VERSION_HEX &gt;= OCC_VERSION_CHECK(7, 4, 0)
    if (format == Format_GLTF)
        return OccGltfReader::createProperties(parentGroup);
    if (format == Format_OBJ)
        return OccObjReader::createProperties(parentGroup);
#endif

    return {};
}

HuguesDelorme, 2021-06-12:
Hello @dansirming 

Sorry for those build problems, though I'm not 100% sure they're caused by me!
What I would like to be sure of is that you don't have any pending change on the Mayo source files that would provoke build errors. Again your git working copy should be "in sync" with the remote `develop` branch.

The Windows/MSVC appveyor CI is the Continuous Integration provided by appveyor for Mayo : each time some changes are pushed to `develop` branch then it triggers a remote build in appveyor(see [last one](https://ci.appveyor.com/project/HuguesDelorme/mayo/builds/39463542))
It's kind of automatic builds, I've configured a "release" build for OpenCascade 7.4.0 and 7.5.0

If the problem still persists then I can propose you to have a phone call next monday through eg Skype so you can screen share the issue.

dansirming, 2021-06-14:
today， I have found  a new problem!  when I imported some steps into mayo ,but it is not in the center of view, it is situated in the left top corner. it is not perfect! it is difficult to use。the test step file in the  mail attachments。can you help me ?

HuguesDelorme, 2021-06-14:
This is an unrelated problem, please create another issue for that and please provide the STEP file

dansirming, 2021-06-19:
ok， thanks！

closed 2022-03-22 by HuguesDelorme