fougue/mayo #67: Mac OS package

livigni, 2021-05-17:
Hi!
Would it be possible to provide a Mac OS package or building instruction?
Thanks.
Ciao

HuguesDelorme, 2021-05-18:
I have to setup a macOS travis build, that's something to be done in the short-term.
In the meanwhile, here is some suggestion(not tested) :

1. Install OpenCascade package
```
brew install opencascade
```

2. Install Qt5 package
```
brew install qt@5
```

3. Build Mayo you need to know where are the lib and include dirs
```
mkdir mayo-build && cd mayo-build
qmake ../mayo CASCADE_INC_DIR=<path> CASCADE_LIB_DIR=<path>
make -j2
```

Schokobecher, 2021-05-24:
I tried my luck with this, but during `make -j4` I get errors like these

> /Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX11.3.sdk/usr/include/c++/v1/filesystem:738:24: note: 'path' has been explicitly marked unavailable here

or
> ~path' is unavailable: introduced in macOS 10.15

and

> 'native' is unavailable: introduced in macOS 10.15

My include and lib dir is

- **/usr/local/Cellar/opencascade/7.5.1_1/include/opencascade/**
- **/usr/local/Cellar/opencascade/7.5.1_1/lib/**

I'm running 11.3.1 Big Sur on an Intel Mac

HuguesDelorme, 2021-10-20:
@m99coder
Obviously the compiler can't find OpenCascade header files.
Can you make sure the path `/usr/local/Cellar/opencascade/7.5.3/include` is correct ?
Please post here the content of that folder(eg `ls -l`)

m99coder, 2021-10-21:
@HuguesDelorme You’re right. Surprisingly inside of `/usr/local/Cellar/opencascade/7.5.3/include` is another folder called `opencascade`.

Now the errors look different. For instance:

```
In file included from ../src/base/application_item.cpp:7:
In file included from ../src/base/application_item.h:9:
In file included from ../src/base/document.h:11:
../src/base/filepath.h:16:35: error: 'path' is unavailable: introduced in macOS 10.15
using FilePath = std::filesystem::path;
                                  ^
/Library/Developer/CommandLineTools/usr/bin/../include/c++/v1/filesystem:738:24: note: 'path' has been explicitly marked unavailable here
class _LIBCPP_TYPE_VIS path {
                       ^
```

Others are:

- ../src/base/filepath.h:19:34: error: 'FilePath' is unavailable: introduced in macOS 10.15
- ../src/base/filepath.h:21:33: error: 'exists' is unavailable: introduced in macOS 10.15
- ../src/base/filepath.h:30:33: error: 'equivalent' is unavailable: introduced in macOS 10.15
- ../src/base/filepath.h:39:33: error: 'is_regular_file' is unavailable: introduced in macOS 10.15

I uninstalled Xcode and only have the command line tools installed (as required by homebrew). Maybe this is the reason?

HuguesDelorme, 2022-05-25:
See [Build instructions for macOS](https://github.com/fougue/mayo/wiki/Build-instructions-for-macOS) wiki page

livigni, 2022-05-25:
> See [Build instructions for macOS](https://github.com/fougue/mayo/wiki/Build-instructions-for-macOS) wiki page

It didn't work for me.
```
vicio@vicio-m1 mayo-build % make -j2
/opt/homebrew/share/qt/libexec/uic ../mayo/src/app/dialog_about.ui -o ui_dialog_about.h
/opt/homebrew/share/qt/libexec/uic ../mayo/src/app/dialog_inspect_xde.ui -o ui_dialog_inspect_xde.h
/opt/homebrew/share/qt/libexec/uic ../mayo/src/app/dialog_options.ui -o ui_dialog_options.h
/opt/homebrew/share/qt/libexec/uic ../mayo/src/app/dialog_save_image_view.ui -o ui_dialog_save_image_view.h
/opt/homebrew/share/qt/libexec/uic ../mayo/src/app/dialog_task_manager.ui -o ui_dialog_task_manager.h
/opt/homebrew/share/qt/libexec/uic ../mayo/src/app/mainwindow.ui -o ui_mainwindow.h
/opt/homebrew/share/qt/libexec/uic ../mayo/src/app/widget_clip_planes.ui -o ui_widget_clip_planes.h
/opt/homebrew/share/qt/libexec/uic ../mayo/src/app/widget_explode_assembly.ui -o ui_widget_explode_assembly.h
/opt/homebrew/share/qt/libexec/uic ../mayo/src/app/widget_model_tree.ui -o ui_widget_model_tree.h
/opt/homebrew/share/qt/libexec/uic ../mayo/src/app/widget_properties_editor.ui -o ui_widget_properties_editor.h
/Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/clang++ -c -pipe -stdlib=libc++ -std=c++17 -Wno-undefined-var-template -O2 -std=gnu++1z  -arch arm64 -isysroot /Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX12.3.sdk -mmacosx-version-min=10.15 -Wall -Wextra -fPIC -DQT_DISABLE_DEPRECATED_BEFORE=0x050F00 -DQT_IMPLICIT_QFILEINFO_CONSTRUCTION -DGL_SILENCE_DEPRECATION -DOCCT_HANDLE_NOCAST -DQT_NO_DEBUG -DQT_OPENGLWIDGETS_LIB -DQT_WIDGETS_LIB -DQT_OPENGL_LIB -DQT_GUI_LIB -DQT_CORE_LIB -I../mayo -I. -Isrc/app -I../mayo/src/app -I../mayo/src/3rdparty -I/usr/local/Cellar/opencascade/7.6.2/include/opencascade -I/opt/homebrew/lib/QtOpenGLWidgets.framework/Headers -I/opt/homebrew/lib/QtWidgets.framework/Headers -I/opt/homebrew/lib/QtOpenGL.framework/Headers -I/opt/homebrew/lib/QtGui.framework/Headers -I/opt/homebrew/lib/QtCore.framework/Headers -I. -I/Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX12.3.sdk/System/Library/Frameworks/OpenGL.framework/Headers -I/Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX12.3.sdk/System/Library/Frameworks/AGL.framework/Headers -I. -I/Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX12.3.sdk/System/Library/Frameworks/OpenGL.framework/Headers -I/Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX12.3.sdk/System/Library/Frameworks/AGL.framework/Headers -I/opt/homebrew/share/qt/mkspecs/macx-clang -F/opt/homebrew/lib -o application.o ../mayo/src/base/application.cpp
/Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/clang++ -c -pipe -stdlib=libc++ -std=c++17 -Wno-undefined-var-template -O2 -std=gnu++1z  -arch arm64 -isysroot /Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX12.3.sdk -mmacosx-version-min=10.15 -Wall -Wextra -fPIC -DQT_DISABLE_DEPRECATED_BEFORE=0x050F00 -DQT_IMPLICIT_QFILEINFO_CONSTRUCTION -DGL_SILENCE_DEPRECATION -DOCCT_HANDLE_NOCAST -DQT_NO_DEBUG -DQT_OPENGLWIDGETS_LIB -DQT_WIDGETS_LIB -DQT_OPENGL_LIB -DQT_GUI_LIB -DQT_CORE_LIB -I../mayo -I. -Isrc/app -I../mayo/src/app -I../mayo/src/3rdparty -I/usr/local/Cellar/opencascade/7.6.2/include/opencascade -I/opt/homebrew/lib/QtOpenGLWidgets.framework/Headers -I/opt/homebrew/lib/QtWidgets.framework/Headers -I/opt/homebrew/lib/QtOpenGL.framework/Headers -I/opt/homebrew/lib/QtGui.framework/Headers -I/opt/homebrew/lib/QtCore.framework/Headers -I. -I/Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX12.3.sdk/System/Library/Frameworks/OpenGL.framework/Headers -I/Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX12.3.sdk/System/Library/Frameworks/AGL.framework/Headers -I. -I/Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX12.3.sdk/System/Library/Frameworks/OpenGL.framework/Headers -I/Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX12.3.sdk/System/Library/Frameworks/AGL.framework/Headers -I/opt/homebrew/share/qt/mkspecs/macx-clang -F/opt/homebrew/lib -o application_item.o ../mayo/src/base/application_item.cpp
In file included from ../mayo/src/base/application_item.cpp:7:
In file included from ../mayo/src/base/application_item.h:9:
In file included from ../mayo/src/base/document.h:9:
../mayo/src/base/application_ptr.h:9:10: fatal error: In file included from ../mayo/src/base/application.cpp:'TDocStd_Application.hxx' file not found7:

In file included from ../mayo/src/base/application.h:9:
../mayo/src/base/application_ptr.h:9:10: fatal error: 'TDocStd_Application.hxx' file not found
#include <TDocStd_Application.hxx>#include <TDocStd_Application.hxx>

1 error generated.
make: *** [application_item.o] Error 1
make: *** Waiting for unfinished jobs....
1 error generated.
make: *** [application.o] Error 1
```

HuguesDelorme, 2022-05-25:
@livigni 
Please post here the output of this command:
````
brew info opencascade
````
This should show where opencascade is located
Thanks for you feedback!

HuguesDelorme, 2022-05-25:
Try those new instructions:
````
mkdir mayo-build
cd mayo-build
CASCADE_CELLAR_DIR=`brew --cellar opencascade`
CASCADE_VERSION=`brew info opencascade | grep -E --only-matching --max-count=1 "[0-9]\.[0-9]\.[0-9]"`
CASCADE_BASE_DIR=$CASCADE_CELLAR_DIR/$CASCADE_VERSION
qmake ../mayo CASCADE_INC_DIR=$CASCADE_BASE_DIR/include/opencascade CASCADE_LIB_DIR=$CASCADE_BASE_DIR/lib
make -j2
````
The install path to the opencascade package is deduced

livigni, 2022-05-25:
> Try those new instructions:
>
> ```
> mkdir mayo-build
> cd mayo-build
> CASCADE_CELLAR_DIR=`brew --cellar opencascade`
> CASCADE_VERSION=`brew info opencascade | grep -E --only-matching --max-count=1 [...]

this worked! thanks

HuguesDelorme, 2022-05-25:
Does Mayo run correctly on your machine ?
I don't have a macOS machine, I know some users previously could build and run Mayo successfully
Would be great if you could post a few screenshots here if you have time

livigni, 2022-05-25:
Here a couple of screenshots. Not sure if the icon of the app is correct (see the dock).
[image]
[image]

HuguesDelorme, 2022-05-30:
Thanks for the screenshots, glad to see Mayo is confirmed to work on macos!
For the application icon I have created #122 some while ago

HuguesDelorme, 2022-09-12:
@livigni 
Seems like the compiler can't find the OpenCascade's header files.
Where is located OpenCascade on your system ?
Make sure the include path is correct.
Also I'm currently having macOS CI issues with homebrew installation of the OpenCascade package.
Maybe this is related

livigni, 2022-09-12:
/opt/homebrew/Cellar/opencascade
version: 7.6.4

HuguesDelorme, 2022-09-12:
What is the result of those commands on your machine:
`brew --cellar opencascade`
and
`brew info opencascade | grep -E --only-matching --max-count=1 "[0-9]\.[0-9]\.[0-9]"`

livigni, 2022-09-13:
vicio@vicio-m1 opencascade % brew --cellar opencascade
/opt/homebrew/Cellar/opencascade

vicio@vicio-m1 opencascade % brew info opencascade | grep -E --only-matching --max-count=1 "[0-9]\.[0-9]\.[0-9]"
7.6.3

HuguesDelorme, 2022-09-13:
Thanks @livigni
Can you make sure the path `/opt/homebrew/Cellar/opencascade/7.6.3/include/opencascade` exists and contains `*.hxx` files ?
What's strange is that you mentionned "version: 7.6.4" in a recent message(yesterday actually)

livigni, 2022-09-13:
yeah, I dunno where that 7.6.4 came from because re-running the command today gave me 7.6.3...
anyway, I think the issue is this: the actual folder path is (see the "_1"). 

I made it work in this way:
mkdir mayo-build
cd mayo-build
CASCADE_CELLAR_DIR=`brew --cellar opencascade`
CASCADE_VERSION=`brew info opencascade | grep -E --only-matching --max-count=1 "[0-9]\.[0-9]\.[0-9]"`
**CASCADE_BASE_DIR=/opt/homebrew/Cellar/opencascade/7.6.3_1/include/opencascade/**
qmake ../mayo CASCADE_INC_DIR=$CASCADE_BASE_DIR/include/opencascade CASCADE_LIB_DIR=$CASCADE_BASE_DIR/lib
make -j2

HuguesDelorme, 2022-09-13:
Hello @livigni 
Good to know you could find the issue and build Mayo with success at the end.
Seems like other people are also experiencing the same issue(see https://github.com/fougue/mayo/issues/151)
Can I ask you to post in this thread the result of the command `brew info opencascade` on your machine ?
I will try to handle that case about specific version(eg 7.6.3_1)
Thanks!

livigni, 2022-09-13:
sure:
vicio@vicio-m1 ~ % brew info opencascade
==> opencascade: stable 7.6.3 (bottled)
3D modeling and numerical simulation software for CAD/CAM/CAE
https://dev.opencascade.org/
/opt/homebrew/Cellar/opencascade/7.6.3_1 (9,457 files, 168.4MB) *
  Poured from bottle on 2022-08-31 at 14:55:33
From: https://github.com/Homebrew/homebrew-core/blob/HEAD/Formula/opencascade.rb
License: LGPL-2.1-only
==> Dependencies
Build: cmake ✘, doxygen ✘, rapidjson ✘
Required: fontconfig ✔, freeimage ✔, freetype ✔, tbb ✔, tcl-tk ✔
==> Analytics
install: 916 (30 days), 1,857 (90 days), 6,382 (365 days)
install-on-request: 538 (30 days), 1,146 (90 days), 4,051 (365 days)
build-error: 3 (30 days)

HuguesDelorme, 2022-09-13:
Thanks @livigni 
That output isn't easy to parse...
Can you please try again but with JSON formatting : `brew info --json opencascade`

livigni, 2022-09-13:
***@***.*** ~ % brew info --json opencascade
[
  {
    "name": "opencascade",
    "full_name": "opencascade",
    "tap": "homebrew/core",
    "oldname": null,
    "aliases": [

    ],
    "versioned_formulae": [

    ],
    "desc": "3D modeling and numerical simulation software for CAD/CAM/CAE",
    "license": "LGPL-2.1-only",
    "homepage": "https://dev.opencascade.org/",
    "versions": {
      "stable": "7.6.3",
      "head": null,
      "bottle": true
    },
    "urls": {
      "stable": {
        "url": "
https://git.dev.opencascade.org/gitweb/?p=occt.git;a=snapshot;h=refs/tags/V7_6_3;sf=tgz
",
        "tag": null,
        "revision": null
      }
    },
    "revision": 1,
    "version_scheme": 0,
    "bottle": {
      "stable": {
        "rebuild": 0,
        "root_url": "https://ghcr.io/v2/homebrew/core",
        "files": {
          "arm64_monterey": {
            "cellar": ":any",
            "url": "
https://ghcr.io/v2/homebrew/core/opencascade/blobs/sha256:3a0ea3b12845cac63ec0adde1f3af3bb079dbde60c37b184fd3b2e00c1c24d32
",
            "sha256":
"3a0ea3b12845cac63ec0adde1f3af3bb079dbde60c37b184fd3b2e00c1c24d32"
          },
          "arm64_big_sur": {
            "cellar": ":any",
            "url": "
https://ghcr.io/v2/homebrew/core/opencascade/blobs/sha256:9f50bbcd16fd80e9da21a2eacc995baa8ec45875a3f2a07dfe9cfb12c26c6cdc
",
            "sha256":
"9f50bbcd16fd80e9da21a2eacc995baa8ec45875a3f2a07dfe9cfb12c26c6cdc"
          },
          "monterey": {
            "cellar": ":any",
            "url": "
https://ghcr.io/v2/homebrew/core/opencascade/blobs/sha256:0ee55bb56dbc02c26f0d6321737ced2ede197f210fc796dc4bfa99c1d422b7ac
",
            "sha256":
"0ee55bb56dbc02c26f0d6321737ced2ede197f210fc796dc4bfa99c1d422b7ac"
          },
          "big_sur": {
            "cellar": ":any",
            "url": "
https://ghcr.io/v2/homebrew/core/opencascade/blobs/sha256:e8c67ba581ab52d35d11c6f7f93fdd582ef3f5185ace8da4b5afd6af743831a4
",
            "sha256":
"e8c67ba581ab52d35d11c6f7f93fdd582ef3f5185ace8da4b5afd6af743831a4"
          },
          "catalina": {
            "cellar": ":any",
            "url": "
https://ghcr.io/v2/homebrew/core/opencascade/blobs/sha256:9b6224a6ccd3484fb4e2b9cd14bfdd3e46ad1b555641f001422fc42cd00eb8be
",
            "sha256":
"9b6224a6ccd3484fb4e2b9cd14bfdd3e46ad1b555641f001422fc42cd00eb8be"
          },
          "x86_64_linux": {
            "cellar": ":any_skip_relocation",
            "url": "
https://ghcr.io/v2/homebrew/core/opencascade/blobs/sha256:b28effcc715e0a7aaeb02ae137536f3326fb540e6115e32602c63cf2a4fd5682
",
            "sha256":
"b28effcc715e0a7aaeb02ae137536f3326fb540e6115e32602c63cf2a4fd5682"
          }
        }
      }
    },
    "keg_only": false,
    "keg_only_reason": null,
    "options": [

    ],
    "build_dependencies": [
      "cmake",
      "doxygen",
      "rapidjson"
    ],
    "dependencies": [
      "fontconfig",
      "freeimage",
      "freetype",
      "tbb",
      "tcl-tk"
    ],
    "recommended_dependencies": [

    ],
    "optional_dependencies": [

    ],
    "uses_from_macos": [

    ],
    "requirements": [

    ],
    "conflicts_with": [

    ],
    "caveats": null,
    "installed": [
      {
        "version": "7.6.3_1",
        "used_options": [

        ],
        "built_as_bottle": true,
        "poured_from_bottle": true,
        "runtime_dependencies": [
          {
            "full_name": "libpng",
            "version": "1.6.37",
            "declared_directly": false
          },
          {
            "full_name": "freetype",
            "version": "2.12.1",
            "declared_directly": true
          },
          {
            "full_name": "fontconfig",
            "version": "2.14.0",
            "declared_directly": true
          },
          {
            "full_name": "freeimage",
            "version": "3.18.0",
            "declared_directly": true
          },
          {
            "full_name": "tbb",
            "version": "2021.5.0",
            "declared_directly": true
          },
          {
            "full_name": "ca-certificates",
            "version": "2022-07-19",
            "declared_directly": false
          },
          {
            "full_name": ***@***.***",
            "version": "1.1.1q",
            "declared_directly": false
          },
          {
            "full_name": "tcl-tk",
            "version": "8.6.12",
            "declared_directly": true
          }
        ],
        "installed_as_dependency": false,
        "installed_on_request": true
      }
    ],
    "linked_keg": "7.6.3_1",
    "pinned": false,
    "outdated": false,
    "deprecated": false,
    "deprecation_date": null,
    "deprecation_reason": null,
    "disabled": false,
    "disable_date": null,
    "disable_reason": null
  }
]

livigni, 2022-09-13:
```
vicio@vicio-m1 ~ % brew info --json opencascade
[
  {
    "name": "opencascade",
    "full_name": "opencascade",
    "tap": "homebrew/core",
    "oldname": null,
    "aliases": [

    ],
    "versioned_formulae": [

    ],
    "desc": "3D modeling and numerical simulation software for CAD/CAM/CAE",
    "license": "LGPL-2.1-only",
    "homepage": "https://dev.opencascade.org/",
    "versions": {
      "stable": "7.6.3",
      "head": null,
      "bottle": true
    },
    "urls": {
      "stable": {
        "url": "https://git.dev.opencascade.org/gitweb/?p=occt.git;a=snapshot;h=refs/tags/V7_6_3;sf=tgz",
        "tag": null,
        "revision": null
      }
    },
    "revision": 1,
    "version_scheme": 0,
    "bottle": {
      "stable": {
        "rebuild": 0,
        "root_url": "https://ghcr.io/v2/homebrew/core",
        "files": {
          "arm64_monterey": {
            "cellar": ":any",
            "url": "https://ghcr.io/v2/homebrew/core/opencascade/blobs/sha256:3a0ea3b12845cac63ec0adde1f3af3bb079dbde60c37b184fd3b2e00c1c24d32",
            "sha256": "3a0ea3b12845cac63ec0adde1f3af3bb079dbde60c37b184fd3b2e00c1c24d32"
          },
          "arm64_big_sur": {
            "cellar": ":any",
            "url": "https://ghcr.io/v2/homebrew/core/opencascade/blobs/sha256:9f50bbcd16fd80e9da21a2eacc995baa8ec45875a3f2a07dfe9cfb12c26c6cdc",
            "sha256": "9f50bbcd16fd80e9da21a2eacc995baa8ec45875a3f2a07dfe9cfb12c26c6cdc"
          },
          "monterey": {
            "cellar": ":any",
            "url": "https://ghcr.io/v2/homebrew/core/opencascade/blobs/sha256:0ee55bb56dbc02c26f0d6321737ced2ede197f210fc796dc4bfa99c1d422b7ac",
            "sha256": "0ee55bb56dbc02c26f0d6321737ced2ede197f210fc796dc4bfa99c1d422b7ac"
          },
          "big_sur": {
            "cellar": ":any",
            "url": "https://ghcr.io/v2/homebrew/core/opencascade/blobs/sha256:e8c67ba581ab52d35d11c6f7f93fdd582ef3f5185ace8da4b5afd6af743831a4",
            "sha256": "e8c67ba581ab52d35d11c6f7f93fdd582ef3f5185ace8da4b5afd6af743831a4"
          },
          "catalina": {
            "cellar": ":any",
            "url": "https://ghcr.io/v2/homebrew/core/opencascade/blobs/sha256:9b6224a6ccd3484fb4e2b9cd14bfdd3e46ad1b555641f001422fc42cd00eb8be",
            "sha256": "9b6224a6ccd3484fb4e2b9cd14bfdd3e46ad1b555641f001422fc42cd00eb8be"
          },
          "x86_64_linux": {
            "cellar": ":any_skip_relocation",
            "url": "https://ghcr.io/v2/homebrew/core/opencascade/blobs/sha256:b28effcc715e0a7aaeb02ae137536f3326fb540e6115e32602c63cf2a4fd5682",
            "sha256": "b28effcc715e0a7aaeb02ae137536f3326fb540e6115e32602c63cf2a4fd5682"
          }
        }
      }
    },
    "keg_only": false,
    "keg_only_reason": null,
    "options": [

    ],
    "build_dependencies": [
      "cmake",
      "doxygen",
      "rapidjson"
    ],
    "dependencies": [
      "fontconfig",
      "freeimage",
      "freetype",
      "tbb",
      "tcl-tk"
    ],
    "recommended_dependencies": [

    ],
    "optional_dependencies": [

    ],
    "uses_from_macos": [

    ],
    "requirements": [

    ],
    "conflicts_with": [

    ],
    "caveats": null,
    "installed": [
      {
        "version": "7.6.3_1",
        "used_options": [

        ],
        "built_as_bottle": true,
        "poured_from_bottle": true,
        "runtime_dependencies": [
          {
            "full_name": "libpng",
            "version": "1.6.37",
            "declared_directly": false
          },
          {
            "full_name": "freetype",
            "version": "2.12.1",
            "declared_directly": true
          },
          {
            "full_name": "fontconfig",
            "version": "2.14.0",
            "declared_directly": true
          },
          {
            "full_name": "freeimage",
            "version": "3.18.0",
            "declared_directly": true
          },
          {
            "full_name": "tbb",
            "version": "2021.5.0",
            "declared_directly": true
          },
          {
            "full_name": "ca-certificates",
            "version": "2022-07-19",
            "declared_directly": false
          },
          {
            "full_name": "openssl@1.1",
            "version": "1.1.1q",
            "declared_directly": false
          },
          {
            "full_name": "tcl-tk",
            "version": "8.6.12",
            "declared_directly": true
          }
        ],
        "installed_as_dependency": false,
        "installed_on_request": true
      }
    ],
    "linked_keg": "7.6.3_1",
    "pinned": false,
    "outdated": false,
    "deprecated": false,
    "deprecation_date": null,
    "deprecation_reason": null,
    "disabled": false,
    "disable_date": null,
    "disable_reason": null
  }
]
```

HuguesDelorme, 2022-09-14:
FYI @livigni https://github.com/fougue/mayo/issues/151#issuecomment-1246426147