fougue/mayo #69: strange looking UI and menu on linux

diplodocuslongus, 2021-05-28:
Hello,

This is an incredible work you have done, well worth the effort to install but I can't seem to have a proper UI.
System: Ubuntu 18.04, gcc 9.3, QT 5.15. 
I've built OpenCascade myself, version 7.5.
I've tried both mayo v0.3.0 and the develop branch.
Here is how it looks on my machine, note the empty square below the cube with the referential and the top menu are cropped (I've shrunken the window height from the default size to make the post smaller).
[image]

Upon launching mayo, I see this in my terminal, not sure if it's relevant.
```
DEBUG: 'opencascade.conf' doesn't exist or is not readable
CRITICAL: "fromVariant() found no enumeration item for ''"
WARNING: QPixmap::scaled: Pixmap is a null pixmap
WARNING: QPixmap::scaled: Pixmap is a null pixmap
```
Knowing that this may be platform specific and not easy to debug, and that linux may not be your platform of choice, I would only ask you if you have any pointer for me to try to debug this issue.

Thanks!

Ludo

HuguesDelorme, 2021-05-28:
Hello Ludo,
Thanks for reporting the issue.
I've just built on Ubuntu 20.04 (VM) with "official" OpenCascade packages (v7.3)
I don't have the icon issue, notice Mayo is using SVG icons so make sure you have the required libs installed :
`sudo apt install libqt5svg5-dev`

I'm also experiencing that CRITICAL error message `fromVariant() found no enumeration item for ''`
And trying to reset options to default values(in Tools->Options) causes a crash so there's obviously an issue that needs to be fixed(will let you know when it's solved)

HuguesDelorme, 2021-05-28:
I've provided in develop branch a related fix for the CRITICAL error message(commit 5f71dd6)
Let me know if it fixes the problem.
Make sure, after compilation, to reset Import/STEP options to default values on first run(Tools->Options)

diplodocuslongus, 2021-05-31:
Hello Hugues,
Thanks. `libqt5svg5-dev` was installed all right but I was still not seeing the icons. After some digging (in the direction you've pointed, that is package / code related to svg), I've found out that what was missing was the package `qt515svg` from the [ppa](https://launchpad.net/%7Ebeineri/+archive/ubuntu/opt-qt-5.15.2-bionic/+index#) I've used to install Qt. So in fact my Qt installation was incomplete, it had nothing to do with mayo. 

I still have some warning messages in the terminal upon launching mayo:
`DEBUG: 'opencascade.conf' doesn't exist or is not readable`
And this warning when opening a file (step , brep, haven't tried all formats):
```
Gtk-Message: 10:01:59.515: GtkDialog mapped without a transient parent. This is discouraged.
TKOpenGl | Type: Other | ID: 0 | Severity: Medium | Message:
  OpenGl_Window::CreateWindow: window Visual is incomplete: no depth buffer, no stencil buffer
```
And finally when I click on the Edit clip planes icon (shown below, I have the error msg only with this icon, the other ones next to it don't show the error message), I see:
`Error: no image library available`
[image]

Since none if these warning and error messages seem to be causing any trouble and since my missing icons issue is solved,  I'm closing this.
Thanks again!

closed 2021-05-31 by diplodocuslongus

HuguesDelorme, 2021-05-31:
@diplodocuslongus 
You can safely ignore this message 
`DEBUG: 'opencascade.conf' doesn't exist or is not readable`

This one is emmited by OpenCascade and I guess it can be ignored as well for now, though it deserves investigation :
`TKOpenGl | Type: Other | ID: 0 | Severity: Medium | Message:
  OpenGl_Window::CreateWindow: window Visual is incomplete: no depth buffer, no stencil buffer`
I also have it on Ubuntu 20.04(running in VM)

But I don't have this message :
`Gtk-Message: 10:01:59.515: GtkDialog mapped without a transient parent. This is discouraged.`
As you said I guess this can be ignored as well