fougue/mayo #70: Very high memory usage with many parts

VictorLamoine, 2021-06-02:
When I try to load a 8 MiB STEP file containing 1947 (simple) parts the memory usage quickly becomes huge and saturates my memory (16 GiB + 2GiB swap)

I can share the file in private but not publicly.

HuguesDelorme, 2021-06-03:
I tried to load your STEP model in "latest" Mayo and it seems to work fine.
With different meshing level here is the memory usage:
Normal: ~180MB
Precise: ~185MB
VeryPrecise: ~200MB

My configuration:
Win10 x64
Visual Studio 2019
OpenCascade v7.5.0 (official binaries from Download Center)
Qt 5.15

Maybe try to reset all your options to default values in Tools->Options ?
If the problem persists let me know

VictorLamoine, 2021-06-03:
My settings were way too precise 😄  Going to "VeryPrecise" the memory usage drops to 360 MiB.

closed 2021-06-03 by VictorLamoine