fougue/mayo #74: Clipping planes: colors

VictorLamoine, 2021-06-10:
When using clip planes on an assembly it's hard to differentiate the parts because all cut surfaces are colored in red:
[image]

What would be nice is to keep the surface colors and ideally add hatching lines like this:
[image]

Example assembly:
https://cad.onshape.com/documents/2ad871b0a5ef2c70b6d5d85f/w/885692aa77d2b4ead62ed359/e/044f794468440e956b6cb205

HuguesDelorme, 2022-05-17:
Done with commit 6dd26ae but still "global" hatching and not per-solid hatching

closed 2022-05-17 by HuguesDelorme

VictorLamoine, 2022-05-22:
Looks much better thanks!
[image]