fougue/mayo #75: Linking Error

devds, 2021-06-18:
I am trying to compile Mayo on my 64bit Windows 10 ,Qt 5.9.9 , OCC 7.5 x64
I have successfully ran the configure script, however when I try to run nmake I get the following error 
 I have also ran vcvars64.bat  before nmake as well in the Visual Studio 17 folder. Any ideas?
I know the problem is the conflict of the two system types (x64/x86) But I don't know how to get passed this problem.
//………………………………………………………………………………
 C:\ mayo-develop\mayo\build>nmake release all
Microsoft (R) Program Maintenance Utility Version 14.16.27045.0
Copyright (C) Microsoft Corporation.  All rights reserved.
"C:\Program Files (x86)\Microsoft Visual Studio\2017\Community\VC\Tools\MSVC\14.16.27023\bin\HostX86\x86\nmake.exe" -f Makefile.Release
Microsoft (R) Program Maintenance Utility Version 14.16.27045.0
Copyright (C) Microsoft Corporation.  All rights reserved.
        link /NOLOGO /DYNAMICBASE /NXCOMPAT /INCREMENTAL:NO /DEBUG /INCREMENTAL:NO /OPT:REF /OPT:ICF /SUBSYSTEM:WINDOWS "/MANIFESTDEPENDENCY:type='win32' name='Microsoft.Windows.Common-Controls' version='6.0.0.0' publicKeyToken='6595b64144ccf1df' language='*' processorArchitecture='*'" /VERSION:0.5 /MANIFEST:embed /OUT:release\mayo.exe @C:\Users\ingma\AppData\Local\Temp\nm5B52.tmp
TKernel.lib(TKernel.dll) : fatal error LNK1112: module machine type 'x64' conflicts with target machine type 'x86'
NMAKE : fatal error U1077: '"C:\Program Files (x86)\Microsoft Visual Studio\2017\Community\VC\Tools\MSVC\14.16.27023\bin\HostX86\x86\link.EXE"' : return code '0x458'
Stop.
NMAKE : fatal error U1077: '"C:\Program Files (x86)\Microsoft Visual Studio\2017\Community\VC\Tools\MSVC\14.16.27023\bin\HostX86\x86\nmake.exe"' : return code '0x2'
Stop.
//………………………………………………………………………..

HuguesDelorme, 2021-06-18:
Hello @devds 
Make sure to follow the [Build instructions for MSVC](https://github.com/fougue/mayo/wiki/Build-instructions-for-Windows-MSVC)

devds, 2021-06-18:
Yes , it's ok, now it runs.
The "explode" function won't run. Moving the slider , no parts explotion occur.I have checked it on a step shape.
Any suggestion?

HuguesDelorme, 2021-06-18:
The exploding is possible only for assemblies, if your part is a single COMPOUND shape then exploding isn't possible
Maybe post a screenshot of the expanded "Model tree" you have in Mayo to see what's your input shape

devds, 2021-06-18:
many thanks

closed 2021-06-18 by HuguesDelorme