fougue/mayo #76: it is not in the center of view, it is situated in the left top corner!

dansirming, 2021-06-19:
today， I have found  a new problem!  when I imported some steps into mayo ,but it is not in the center of view, it is situated in the left top corner. it is not perfect! it is difficult to use。the test step file in the  mail attachments。can you help me ?
[888.zip](https://github.com/fougue/mayo/files/6680105/888.zip)

HuguesDelorme, 2021-06-20:
The problem is caused by the part in 888.step being located far in X negative (around -1250) and the 3D origin trihedron is taken into account in the "fit all" action (it should not).
As a temporary workaround you can disable this option :
[image]

So each time you load a file the 3D origin trihedron will be hidden by default so the "fit all" action will behave as you expect.

HuguesDelorme, 2023-01-12:
Acceptable workaround exists

closed 2023-01-12 by HuguesDelorme