fougue/mayo #82: Compile error: redefinition of 'StringConv<....>' in string_conv.h

abrock, 2021-08-05:
I get the following compile error:

`VERBOSE=1 make
g++ -c -pipe -std=c++17 -O2 -std=gnu++1z -Wall -Wextra -D_REENTRANT -fPIC -DHAVE_CONFIG_H -DHAVE_FSTREAM -DHAVE_IOSTREAM -DHAVE_IOMANIP -DHAVE_LIMITS_H -DOCCT_HANDLE_NOCAST -DLIN -DLININTEL -DOCC_CONVERT_SIGNALS -D_OCC64 -DQT_NO_DEBUG -DQT_WIDGETS_LIB -DQT_GUI_LIB -DQT_CORE_LIB -I../mayo -I. -Isrc/app -I../mayo/src/app -I../mayo/src/3rdparty -I/usr/include/opencascade -I/usr/include/x86_64-linux-gnu/qt5 -I/usr/include/x86_64-linux-gnu/qt5/QtWidgets -I/usr/include/x86_64-linux-gnu/qt5/QtGui -I/usr/include/x86_64-linux-gnu/qt5/QtCore -I. -I. -I/usr/lib/x86_64-linux-gnu/qt5/mkspecs/linux-g++ -o occ_progress_indicator.o ../mayo/src/base/occ_progress_indicator.cpp
In file included from ../mayo/src/base/occ_progress_indicator.cpp:8:
../mayo/src/base/string_conv.h:137:19: error: redefinition of ‘struct Mayo::StringConv<std::__cxx11::basic_string<char>, opencascade::handle<TCollection_HAsciiString> >’
  137 | template<> struct StringConv<std::string, opencascade::handle<TCollection_HAsciiString>> {
      |                   ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
../mayo/src/base/string_conv.h:89:19: note: previous definition of ‘struct Mayo::StringConv<std::__cxx11::basic_string<char>, opencascade::handle<TCollection_HAsciiString> >’
   89 | template<> struct StringConv<std::string, Handle_TCollection_HAsciiString> {
      |                   ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
../mayo/src/base/string_conv.h:145:19: error: redefinition of ‘struct Mayo::StringConv<QString, opencascade::handle<TCollection_HAsciiString> >’
  145 | template<> struct StringConv<QString, opencascade::handle<TCollection_HAsciiString>> {
      |                   ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
../mayo/src/base/string_conv.h:121:19: note: previous definition of ‘struct Mayo::StringConv<QString, opencascade::handle<TCollection_HAsciiString> >’
  121 | template<> struct StringConv<QString, Handle_TCollection_HAsciiString> {
      |                   ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
../mayo/src/base/string_conv.h:153:19: error: redefinition of ‘struct Mayo::StringConv<opencascade::handle<TCollection_HAsciiString>, QString>’
  153 | template<> struct StringConv<opencascade::handle<TCollection_HAsciiString>, QString> {
      |                   ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
../mayo/src/base/string_conv.h:129:19: note: previous definition of ‘struct Mayo::StringConv<opencascade::handle<TCollection_HAsciiString>, QString>’
  129 | template<> struct StringConv<Handle_TCollection_HAsciiString, QString> {
      |                   ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
make: *** [Makefile:2020: occ_progress_indicator.o] Error 1
`

If I remove lines 135-158 in string_conv.h it compiles and works as expected, I can open STL and STEP files and rotate them.

I run Debian bullseye, "Linux 5.10.0-8-amd64 #1 SMP Debian 5.10.46-3 (2021-07-28) x86_64 GNU/Linux".
"g++ --version" says "g++ (Debian 10.2.1-6) 10.2.1 20210110"

HuguesDelorme, 2021-08-06:
@abrock 
Does a6a2e49326659694ec1ee80702ff51fe088990ac fix the compilation errors you reported ?
Thanks for your feedback

abrock, 2021-08-06:
Yes, it compiles and works as expected now.

closed 2021-08-06 by abrock

abrock, 2021-08-06:
Thank you for providing such a nice 3D viewer.