fougue/mayo #94: QStringView not found

karimnorouzi, 2021-12-15:
When I try to compile the code in qtcreator, it complains that qstringview file not found.
Though strictly not a mayo code error, I encountred this error here.

HuguesDelorme, 2021-12-15:
What is the version of Qt you are using ?
QStringView was introduced with Qt 5.10 (see https://doc.qt.io/qt-5/qstringview.html)

karimnorouzi, 2021-12-15:
qt 5.9.9 msvc 2019.
I tried to compile the program inside qtcreator if it is any help.

HuguesDelorme, 2021-12-15:
That's the problem : the Qt version you are using is just a bit too old.
You can try to comment lines L65 to L70 in file `src/app/qstring_conv.h` and compile again.
Or switch to latest version of Qt5 (should be 5.15.x)

karimnorouzi, 2021-12-15:
why 5.9.9 is older than 5.15 ?

HuguesDelorme, 2021-12-15:
With regard to [semantic versioning](https://semver.org) Qt 5.9 is older than Qt 5.15
The third number is the "patch" version

karimnorouzi, 2021-12-15:
Oh.
I got it.
thanks.

closed 2021-12-15 by HuguesDelorme