fougue/mayo #97: HiDPI

VictorLamoine, 2022-02-04:
On high resolution screens the application does not display correctly:

Here on a laptop with a 2880x1800 screen resolution:
[image]

3D viewer is too small

Running `QT_SCALE_FACTOR=0.5 ./mayo`:
[image]

Now the viewer is correctly sized but the text is very small.

HuguesDelorme, 2022-02-04:
Thanks for reporting!
Duplicates #54

HuguesDelorme, 2023-02-10:
@VictorLamoine 
Branch bugfix/highdpi is an attempt to fix high DPI support.
I don't have a suitable screen for real testing, I used QT_SCALE_FACTOR to simulate higher DPI.
It would be great if you have some time to check that branch works on your local machine!
Let me know any feedback
Greetings

VictorLamoine, 2023-03-02:
Sorry for the late reply: It now works perfectly out of the box on a HDPI screen, good work 👍🏼

HuguesDelorme, 2023-03-02:
Good news!
Many thanks for your feedback

closed 2023-03-02 by HuguesDelorme