fougue/mayo #100: mayo on aarch64

KascheyBessmertnii, 2022-02-15:
Hello!

I tried to build this project for a processor based on ARM64 architecture for Linux (armv8, cortex A57, aarch64)

When trying to build i have an error which says "Platform architecture not supported (QT_ARCH=arm64)"

Could you specify is there a version suitable for building on ARM64 archtecture?
Or do you have any ideas at which direction to look to solve it? 
I didn't have experience working with !t before, so maybe its some simple configs somewhere in the project

HuguesDelorme, 2022-02-15:
Hello Kaschey!
Thanks for having reported this issue. 
This is a somewhat "forced" error that is located in file [opencascade.pri](https://github.com/fougue/mayo/blob/0155cc5edf15e13dcbae156e5b5b8c94b55eaf15/opencascade.pri#L50)
This is too restrictive, I should turn this `error()` statement into a `warning()`
Try to remove that line (or comment it with #) and run qmake again.
Let me know the results of this, that's interesting.
By the way did you manage to install OpenCascade packages on your Linux system ?

KascheyBessmertnii, 2022-02-15:
Thanks for the tip,
Changing error to warning helped to install Mayo on Alt Linux 10 (aarch64) on armv8-a processor.
I also had to add in method 

QStringQStringUtils::yesNoText(CheckState state)  ( mayo/src/app/qstring_utils.cpp ) 
default: return tr("No");

After this it compiled. But when i tried to open step file (tried on other PCs with mayo it worked great) around 20 megabytes an error occures.
A branch treee of elements works good, but the model itself doesnt render

HuguesDelorme, 2022-02-15:
At least it compiles that's a good point.
What is the Qt and OpenCascade versions you are using ?
What C++ compiler ?

KascheyBessmertnii, 2022-02-15:
Qt - 5.15.2
OpenCascade - 7.5.3
gcc - 10.3.1

HuguesDelorme, 2022-02-15:
No problem with that configuration then.
If you run debug build of mayo don't you see any OpenGL warning in the console output ?

HuguesDelorme, 2022-04-12:
Hello @CheLoV
Did you try to run some official OpenCascade example on your arm64 OS ?

HuguesDelorme, 2023-04-04:
@KascheyBessmertnii and @CheLoV
Do you still experience this problem with latest version(v0.7.0)?
Let me know your feedback if any, thanks