fougue/mayo #109: black screen when open file at version 0.5.1

younghang, 2022-03-03:
[image]
When I am using version 0.4 everything is OK, I got black screen when I update the mayo to version 0.5.1. I can't see anything in the model window. I don't know what is wrong.

HuguesDelorme, 2022-03-03:
Fixed with commit https://github.com/fougue/mayo/commit/3510c932d9e31a9e7d2f160874ed42b42d13759e

closed 2022-03-03 by HuguesDelorme

HuguesDelorme, 2022-03-03:
@younghang
Please install bugfix release v0.5.2 and let me know if that problem is solved on your side
https://github.com/fougue/mayo/releases/download/v0.5.2/mayo_v0.5.2.860-2f9229e_winx64_installer.exe

younghang, 2022-03-04:
@HuguesDelorme It has been solved. Thanks. But I feel it becomes less fluent, maybe it is my illusion.

HuguesDelorme, 2022-03-04:
@younghang
To verify the potential slowdown, you can activate performance stats by clicking on menu "Display->Show Performance Stats" and check the FPS
If you have both 0.4 and 0.5.2 versions installed it would be interesting to compare.