fougue/mayo #112: Link error

yvanblanchard, 2022-03-08:
Hello,

I have some link errors (see image below).
I am using:
Qt Creator 4.133.2, with QT v5.13.2, and MSVC2017 x64.
Open-Cascade v7.6.0.

[image]

Could you please help ?
Thank you.

HuguesDelorme, 2022-03-09:
Are you trying to build latest commit in develop branch?
I can't see what could cause these errors as the appveyor CI config is nearly the same as yours.
See:
https://github.com/fougue/mayo/blob/develop/appveyor.yml
https://ci.appveyor.com/project/HuguesDelorme/mayo/build/job/o1ju2kbc0rxhep5i

I would recommend to clean output build directory and run QMake step again.

HuguesDelorme, 2022-03-14:
@yvanblanchard did you manage to solve those errors ?

yvanblanchard, 2022-03-14:
@HuguesDelorme Unfortunately, I still have those linking errors, even after cleaning the build directory..

HuguesDelorme, 2022-03-15:
There's certainly an issue with your git clone because `io_occ_obj.cpp` is no longer in Mayo codebase.
Try to run "git clean -dxf"
If the error persists then I would recommend to clone Mayo again

yvanblanchard, 2022-03-16:
Hello
Thank you. Cloning the repo again solved the issue

closed 2022-03-16 by yvanblanchard