fougue/mayo #113: build error: cannot convert ‘std::nullptr_t’ to ‘Aspect_Drawable’

anverx, 2022-03-10:
Strictly following the build instructions on gentoo distro

```
*  sci-libs/opencascade
      Latest version available: 7.6.0-r2
      Latest version installed: 7.6.0-r2
*  dev-qt/qtsvg
      Latest version available: 5.15.2-r13
      Latest version installed: 5.15.2-r13
*  sys-devel/gcc
      Latest version available: 11.2.0
      Latest version installed: 11.2.0
```

And the error:
```
g++ -c -pipe -std=c++17 -O2 -std=gnu++1z -Wall -Wextra -D_REENTRANT -fPIC -DQT_DISABLE_DEPRECATED_BEFORE=0x050F00 -DHAVE_CONFIG_H -DHAVE_FSTREAM -DHAVE_IOSTREAM -DHAVE_IOMANIP -DHAVE_LIMITS_H -DOCCT_HANDLE_NOCAST -DLIN -DLININTEL -DOCC_CONVERT_SIGNALS -D_OCC64 -DQT_NO_DEBUG -DQT_WIDGETS_LIB -DQT_GUI_LIB -DQT_CORE_LIB -I../mayo -I. -Isrc/app -I../mayo/src/app -I../mayo/src/3rdparty -I/usr/include/opencascade -I/usr/include/qt5 -I/usr/include/qt5/QtWidgets -I/usr/include/qt5/QtGui -I/usr/include/qt5/QtCore -I. -I. -I/usr/lib64/qt5/mkspecs/linux-g++ -o widget_properties_editor.o ../mayo/src/app/widget_properties_editor.cpp
../mayo/src/app/widget_occ_view.cpp: In member function ‘virtual void Mayo::QOpenGLWidgetOccView::initializeGL()’:
../mayo/src/app/widget_occ_view.cpp:136:37: error: cannot convert ‘std::nullptr_t’ to ‘Aspect_Drawable’ {aka ‘long unsigned int’} in initialization
  136 |         Aspect_Drawable nativeWin = nullptr;
      |                                     ^~~~~~~
make: *** [Makefile:3874: widget_occ_view.o] Error 1
```

HuguesDelorme, 2022-03-10:
@anverx
Thanks for reporting this compilation error!
Could you please try this fix and let me know:
`136 |         Aspect_Drawable nativeWin = 0;`

The cause is that `Aspect_Drawable` is alias to `void*` on Windows and `unsigned long` elsewhere so it's a different type depending on the platform. And there is another degree of variation with OpenCascade version. The current travis CI for Linux is using OpenCascade v7.3 and does not compile this code(because `QOpenGLWidgetOccView` is available only with version >= 7.6)

anverx, 2022-03-11:
`Aspect_Drawable nativeWin = 0;`  fixes compilation.  Next problem is linking
```
g++ -Wl,-O1 -o mayo application.o application_item.o application_item_selection_model.o bnd_utils.o brep_utils.o caf_utils.o document.o document_tree_node.o document_tree_node_properties_provider.o enumeration.o geom_utils.o io_format.o io_reader.o io_system.o io_writer.o math_utils.o mesh_utils.o messenger.o occ_progress_indicator.o occ_static_variables_rollback.o occt_enums.o property.o property_builtins.o property_enumeration.o property_value_conversion.o quantity.o settings.o task.o task_manager.o task_progress.o text_id.o tkernel_utils.o unit.o unit_system.o xcaf.o io_occ.o io_occ_base_mesh.o io_occ_brep.o io_occ_caf.o io_occ_common.o io_occ_gltf_reader.o io_occ_gltf_writer.o io_occ_iges.o io_occ_obj_reader.o io_occ_obj_writer.o io_occ_step.o io_occ_stl.o io_occ_vrml.o dxf.o io_dxf.o io_image.o ais_text.o graphics_create_driver.o graphics_create_virtual_window.o graphics_mesh_data_source.o graphics_object_base_property_group.o graphics_object_driver.o graphics_object_driver_table.o graphics_scene.o graphics_tree_node_mapping.o graphics_tree_node_mapping_driver.o graphics_tree_node_mapping_driver_table.o graphics_utils.o v3d_view_camera_animation.o v3d_view_controller.o gui_application.o gui_document.o gui_document_list_model.o qtgui_utils.o app_module.o button_flat.o console.o dialog_about.o dialog_inspect_xde.o dialog_options.o dialog_save_image_view.o dialog_task_manager.o document_property_group.o document_tree_node_properties_providers.o grid_helper.o item_view_buttons.o list_helper.o main.o mainwindow.o occt_window_740.o occt_window_750.o property_editor_factory.o property_item_delegate.o proxy_styled_item_delegate.o qsettings_storage.o qstring_utils.o recent_files.o theme.o widget_clip_planes.o widget_explode_assembly.o widget_file_system.o widget_gui_document.o widget_home_files.o widget_message_indicator.o widget_model_tree.o widget_model_tree_builder.o widget_model_tree_builder_mesh.o widget_model_tree_builder_xde.o widget_occ_view.o widget_occ_view_controller.o widget_properties_editor.o widgets_utils.o format.o qrc_mayo.o moc_application.o moc_application_item_selection_model.o moc_document.o moc_property.o moc_settings.o moc_task_manager.o moc_graphics_object_base_property_group.o moc_graphics_scene.o moc_v3d_view_controller.o moc_gui_application.o moc_gui_document.o moc_app_module.o moc_button_flat.o moc_dialog_about.o moc_dialog_inspect_xde.o moc_dialog_options.o moc_dialog_save_image_view.o moc_dialog_task_manager.o moc_item_view_buttons.o moc_list_helper.o moc_mainwindow.o moc_property_item_delegate.o moc_widget_clip_planes.o moc_widget_explode_assembly.o moc_widget_file_system.o moc_widget_gui_document.o moc_widget_home_files.o moc_widget_message_indicator.o moc_widget_model_tree.o moc_widget_occ_view_controller.o   -lTKBin -lTKBinL -lTKBinXCAF -lTKBO -lTKBool -lTKBRep -lTKCAF -lTKCDF -lTKernel -lTKG2d -lTKG3d -lTKGeomAlgo -lTKGeomBase -lTKHLR -lTKLCAF -lTKMath -lTKMesh -lTKMeshVS -lTKOpenGl -lTKPrim -lTKService -lTKShHealing -lTKTopAlgo -lTKXSDRAW -lTKV3d -lTKVCAF -lTKXCAF -lTKXml -lTKXmlL -lTKXmlXCAF -lTKXSBase -lTKIGES -lTKXDEIGES -lTKSTEP -lTKSTEP209 -lTKSTEPAttr -lTKSTEPBase -lTKXDESTEP -lTKSTL -lTKRWMesh -lTKVRML /usr/lib64/libQt5Widgets.so /usr/lib64/libQt5Gui.so /usr/lib64/libQt5Core.so -lGL -lpthread   
/usr/lib/gcc/x86_64-pc-linux-gnu/11.2.0/../../../../x86_64-pc-linux-gnu/bin/ld: cannot find -lTKBin
/usr/lib/gcc/x86_64-pc-linux-gnu/11.2.0/../../../../x86_64-pc-linux-gnu/bin/ld: cannot find -lTKBinL
/usr/lib/gcc/x86_64-pc-linux-gnu/11.2.0/../../../../x86_64-pc-linux-gnu/bin/ld: cannot find -lTKBinXCAF
/usr/lib/gcc/x86_64-pc-linux-gnu/11.2.0/../../../../x86_64-pc-linux-gnu/bin/ld: cannot find -lTKBO
/usr/lib/gcc/x86_64-pc-linux-gnu/11.2.0/../../../../x86_64-pc-linux-gnu/bin/ld: cannot find -lTKBool
/usr/lib/gcc/x86_64-pc-linux-gnu/11.2.0/../../../../x86_64-pc-linux-gnu/bin/ld: cannot find -lTKBRep
/usr/lib/gcc/x86_64-pc-linux-gnu/11.2.0/../../../../x86_64-pc-linux-gnu/bin/ld: cannot find -lTKCAF
/usr/lib/gcc/x86_64-pc-linux-gnu/11.2.0/../../../../x86_64-pc-linux-gnu/bin/ld: cannot find -lTKCDF
/usr/lib/gcc/x86_64-pc-linux-gnu/11.2.0/../../../../x86_64-pc-linux-gnu/bin/ld: cannot find -lTKernel
/usr/lib/gcc/x86_64-pc-linux-gnu/11.2.0/../../../../x86_64-pc-linux-gnu/bin/ld: cannot find -lTKG2d
/usr/lib/gcc/x86_64-pc-linux-gnu/11.2.0/../../../../x86_64-pc-linux-gnu/bin/ld: cannot find -lTKG3d
/usr/lib/gcc/x86_64-pc-linux-gnu/11.2.0/../../../../x86_64-pc-linux-gnu/bin/ld: cannot find -lTKGeomAlgo
/usr/lib/gcc/x86_64-pc-linux-gnu/11.2.0/../../../../x86_64-pc-linux-gnu/bin/ld: cannot find -lTKGeomBase
/usr/lib/gcc/x86_64-pc-linux-gnu/11.2.0/../../../../x86_64-pc-linux-gnu/bin/ld: cannot find -lTKHLR
/usr/lib/gcc/x86_64-pc-linux-gnu/11.2.0/../../../../x86_64-pc-linux-gnu/bin/ld: cannot find -lTKLCAF
/usr/lib/gcc/x86_64-pc-linux-gnu/11.2.0/../../../../x86_64-pc-linux-gnu/bin/ld: cannot find -lTKMath
/usr/lib/gcc/x86_64-pc-linux-gnu/11.2.0/../../../../x86_64-pc-linux-gnu/bin/ld: cannot find -lTKMesh
/usr/lib/gcc/x86_64-pc-linux-gnu/11.2.0/../../../../x86_64-pc-linux-gnu/bin/ld: cannot find -lTKMeshVS
/usr/lib/gcc/x86_64-pc-linux-gnu/11.2.0/../../../../x86_64-pc-linux-gnu/bin/ld: cannot find -lTKOpenGl
/usr/lib/gcc/x86_64-pc-linux-gnu/11.2.0/../../../../x86_64-pc-linux-gnu/bin/ld: cannot find -lTKPrim
[... cut ...]
/usr/lib/gcc/x86_64-pc-linux-gnu/11.2.0/../../../../x86_64-pc-linux-gnu/bin/ld: cannot find -lTKXDESTEP
/usr/lib/gcc/x86_64-pc-linux-gnu/11.2.0/../../../../x86_64-pc-linux-gnu/bin/ld: cannot find -lTKSTL
/usr/lib/gcc/x86_64-pc-linux-gnu/11.2.0/../../../../x86_64-pc-linux-gnu/bin/ld: cannot find -lTKRWMesh
/usr/lib/gcc/x86_64-pc-linux-gnu/11.2.0/../../../../x86_64-pc-linux-gnu/bin/ld: cannot find -lTKVRML
collect2: error: ld returned 1 exit status
make: *** [Makefile:681: mayo] Error 1
```
on my system these libs are in `/usr/lib64/opencascade/`

HuguesDelorme, 2022-03-12:
On Debian-based systems when official OpenCascade are used there's no need to specfify the libs path.
On Gentoo that's obviously different, I suggest you run again the QMake step:
````
$> qmake ../mayo CASCADE_INC_DIR=/usr/include/opencascade CASCADE_LIB_DIR=/usr/lib64/opencascade
$> make -j2
````

anverx, 2022-03-14:
:+1:  compiles fine!
(crashes when loading file but that's because my opengl is misconfigured, gotta get that fixed first)

closed 2022-03-14 by HuguesDelorme

HuguesDelorme, 2022-03-24:
Thanks for your feedback!