fougue/mayo #117: Crash at startup

Homo-carbonis, 2022-04-01:
I built Mayo against openCascade 7.4.0 and it is segfaulting at startup.
```
Program terminated with signal SIGSEGV, Segmentation fault.
#0  0x000055a19bc22c72 in Mayo::PropertyEnumeration::PropertyEnumeration(Mayo::PropertyGroup*, Mayo::TextId const&, Mayo::Enumeration const&) ()
(gdb) bt
#0  0x000055a19bc22c72 in Mayo::PropertyEnumeration::PropertyEnumeration(Mayo::PropertyGroup*, Mayo::TextId const&, Mayo::Enumeration const&) ()
#1  0x000055a19bc30bb3 in Mayo::IO::OccBaseMeshReaderProperties::OccBaseMeshReaderProperties(Mayo::PropertyGroup*) ()
#2  0x000055a19bc380c9 in Mayo::IO::OccObjReader::createProperties(Mayo::PropertyGroup*) ()
#3  0x000055a19bc2f885 in Mayo::IO::OccFactoryReader::createProperties(Mayo::IO::Format, Mayo::PropertyGroup*) const ()
#4  0x000055a19bc871cf in Mayo::AppModule::AppModule(Mayo::Application*) ()
#5  0x000055a19bcabb4e in Mayo::runApp(QCoreApplication*) ()
#6  0x000055a19bbfc14d in main ()
```

HuguesDelorme, 2022-04-01:
Hello thanks for reporting this issue
This is recent regression detected in the appveyor CI
I'm going to provide a fix, will let you know when it's done

HuguesDelorme, 2022-04-01:
@Homo-carbonis 
Fixed with 3a28b947d661499f5897e888bce9a0d3a66b2981

closed 2022-04-01 by HuguesDelorme