fougue/mayo #118: Crash in docker container ubuntu 20

joshco, 2022-04-16:
Hi, 
I'm trying to run mayo in a docker container, and then export my DISPLAY back to an X server.  I'm getting the following error on startup, any ideas?

> terminate called after throwing an instance of 'std::out_of_range'
  what():  vector::_M_range_check: __n (which is 0) >= this->size() (which is 0)

Versions
root@7d3239eaf4b8:/mayo-build# qmake --version
QMake version 3.1
Using Qt version 5.12.8 in /usr/lib/x86_64-linux-gnu
OpenCascade  7.3.3+dfsg1-1build1

Here is a stack trace:
root@7d3239eaf4b8:/mayo-build# gdb ./mayo
GNU gdb (Ubuntu 9.2-0ubuntu1~20.04.1) 9.2
Copyright (C) 2020 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.
Type "show copying" and "show warranty" for details.
This GDB was configured as "x86_64-linux-gnu".
Type "show configuration" for configuration details.
For bug reporting instructions, please see:
<http://www.gnu.org/software/gdb/bugs/>.
Find the GDB manual and other documentation resources online at:
    <http://www.gnu.org/software/gdb/documentation/>.

For help, type "help".
Type "apropos word" to search for commands related to "word"...
Reading symbols from ./mayo...
(No debugging symbols found in ./mayo)
(gdb) run
Starting program: /mayo-build/mayo
warning: Error disabling address space randomization: Operation not permitted
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib/x86_64-linux-gnu/libthread_db.so.1".
[New Thread 0x7f52b631e700 (LWP 346)]
WARNING: QStandardPaths: XDG_RUNTIME_DIR not set, defaulting to '/tmp/runtime-root'
terminate called after throwing an instance of 'std::out_of_range'
  what():  vector::_M_range_check: __n (which is 0) >= this->size() (which is 0)

Thread 1 "mayo" received signal SIGABRT, Aborted.
__GI_raise (sig=sig@entry=6) at ../sysdeps/unix/sysv/linux/raise.c:50
50      ../sysdeps/unix/sysv/linux/raise.c: No such file or directory.
(gdb) where
#0  __GI_raise (sig=sig@entry=6) at ../sysdeps/unix/sysv/linux/raise.c:50
#1  0x00007f52bb3d8859 in __GI_abort () at abort.c:79
#2  0x00007f52bb7b2911 in ?? () from /lib/x86_64-linux-gnu/libstdc++.so.6
#3  0x00007f52bb7be38c in ?? () from /lib/x86_64-linux-gnu/libstdc++.so.6
#4  0x00007f52bb7be3f7 in std::terminate() () from /lib/x86_64-linux-gnu/libstdc++.so.6
#5  0x00007f52bb7be6a9 in __cxa_throw () from /lib/x86_64-linux-gnu/libstdc++.so.6
#6  0x00007f52bb7b53ab in ?? () from /lib/x86_64-linux-gnu/libstdc++.so.6
#7  0x000056151595f1d2 in Mayo::PropertyEnumeration::PropertyEnumeration(Mayo::PropertyGroup*, Mayo::TextId const&, Mayo::Enumeration const*) ()
#8  0x000056151599ddff in Mayo::IO::DxfReader::Properties::Properties(Mayo::PropertyGroup*) ()
#9  0x00005615159971e7 in Mayo::IO::DxfReader::createProperties(Mayo::PropertyGroup*) ()
#10 0x000056151599724d in Mayo::IO::DxfFactoryReader::createProperties(Mayo::IO::Format, Mayo::PropertyGroup*) const ()
#11 0x00005615159c3af3 in Mayo::AppModule::AppModule(Mayo::Application*) ()
#12 0x00005615159ef45c in Mayo::runApp(QCoreApplication*) ()
#13 0x00005615159369a2 in main ()
(gdb)

HuguesDelorme, 2022-04-19:
Hello @joshco 

It seems like you're not using latest version, because looking at stack trace it reminds me of something I fixed a while ago.
See commit https://github.com/fougue/mayo/commit/3a28b947d661499f5897e888bce9a0d3a66b2981
Make sure you build latest version from develop branch

HuguesDelorme, 2022-04-19:
Might be a duplicate of #117

HuguesDelorme, 2022-09-13:
No activity on this issue

closed 2022-09-13 by HuguesDelorme