fougue/mayo #120: Possibility to use Mayo as lib instead of command-line for converting formats

yvanblanchard, 2022-04-24:
Hello,

I would like to know if it’s already possible to call your code as a (static) library for importing and exporting , instead of executable command line, as silent mode (no GUI). Or if additional code would be required .
If yes (already feasible ), could you give a sample code for implementing the lib.

thank you.

HuguesDelorme, 2022-04-25:
Hello @yvanblanchard 
That should be perfectly possible.
Have a look at that [gist](https://gist.github.com/HuguesDelorme/b9eddc7f9adec5cef61bb04ef1724024)
It provides a QMake project file that builds base/ and io/ source folders as a static library.
There's also a main.cpp example to load/convert files(just for demo: no progress, no error handling, ...)
Warning: not tested!

closed 2022-05-17 by HuguesDelorme

Abde5, 2022-09-19:
It would be interesting to bypass Qt when compiling as a library. Is it really a dependency when we use it as a CLI/lib, apart from using it for signaling?

HuguesDelorme, 2022-09-19:
Hello @Abde5 
Yes you're right, for the source code in `base/` and `io/` Qt is only used for signals
In the past I removed other usage of Qt but kept only signals
They could be replaced with the [KDBindings](https://github.com/KDAB/KDBindings) library

See this [new task](https://github.com/fougue/mayo/issues/154) to cover that specific topic

Abde5, 2022-09-26:
Hello @HuguesDelorme

Thanks again for the new issue. I am contacting you because I am having some problems using Mayo as a standalone library.
I used the main file that you provided with some adjustments, and everything seems to work. However, trying to export a STEP file
to binary glTF (the default parameters) gives a 250MB file while the export from the main application gives a 20MB file.

Also, OBJ files do not seem to be imported with the .mat files - again, this does not happen in the graphical interface. 
I think that this problem is because of a lack of understanding on how to configure the `Reader` classes. However, I saw that the 
default parameters are configured in `io_occ_*.h` with the same values as in the main application.

May you please give me a tip on where the problem could be? Is it maybe the configuration of OCC?

Thanks again for your work.

HuguesDelorme, 2022-09-26:
Hello @Abde5 
Maybe this is related to how you transfer the data to glTF writer.
Are you passing the whole Document object ?
If it doesn't help then share a gist with an excerpt of your code

Abde5, 2022-09-26:
I am indeed passing the whole document to export.

However, I do not think that it is a problem of the writer. The `ImportInDocument` call uses a lot of resources while reading the file with entity post process. While it takes somes seconds to import in the GUI, it takes some minutes to import in the "lib" version.

[Here](https://gist.github.com/Abde5/c942db4329c929f10a2fc9206a23d885) you can find a gist with the main file that I am using. You can notice that it's quite similar to the one you shared before.

HuguesDelorme, 2022-09-26:
Well spotted, have a look at this [new gist](https://gist.github.com/HuguesDelorme/7a31da6e217f667c2af553b5debcfa89)
It introduces meshing of the BREP shapes.
That was something missing(intended) in my original gist.

Abde5, 2022-09-26:
Thank you so much @HuguesDelorme. It is working well now, the import process is much faster 
and looks like it goes as well as the GUI.

However, I am still facing some problems. Some step files seem to be imported with corrupted geometry as glb. Also, the `mergeFaces` of the GltfWriter does not seem to have any effect in the output file. 

I will try to research this. There is maybe something I did wrong. 

Thanks again for spending time in this!