fougue/mayo #123: CATIA mouse settings

yvanblanchard, 2022-04-30:
Hello,

could it be possible to use CATIA V5 settings for mouse (pan, rotate, rotation center,..) ? It would be very useful.

Thank you.

HuguesDelorme, 2022-05-01:
Yes that should be rather easy to implement.
Can you please provide here the detailed mouse/keyboard settings to control the 3D view in CATIA ?

yvanblanchard, 2022-05-01:
That’s a good new.
Here is a link to the Abaqus cae documentation, with view mouse settings description (including catia and solid works).
Thank you again, your software is very nice and helpful.

HuguesDelorme, 2022-05-05:
Duplicated by #125

HuguesDelorme, 2022-05-05:
@yvanblanchard 
Can you please test latest Mayo and report feedback regarding this feature?

closed 2022-05-05 by HuguesDelorme

yvanblanchard, 2022-05-05:
I just tested and it works nice !
Thank you. Great improvement !