fougue/mayo #126: Command-line Mayo: file conversion settings

yvanblanchard, 2022-05-10:
Hello,

When using MAYO through command-line, for converting files: is it possible to adjust some settings (ex: tesselation accuracy, etc).
It seems that when converting a STEP assembly to STL mesh file, result was not so good (big triangles, etc), compared to same conversion bu through GUI (with default settings).

thank you.

HuguesDelorme, 2022-05-10:
Yes it's possible to override settings through a custom INI file with `--settings myconfig.ini` on the command-line.

1. In Mayo open the Options dialog with Tools->Option menu
2. Change the "BRep Meshing" options to increase accuracy(eg set Quality option to "Precise" or "VeryPrecise")
3. Click the "Exchange" button and select "Save as..." in the drop-down menu
4. Select and validate a location for the INI file to be exported(eg. `myconfig.ini`)
5. Click the "Cancel" button of the Options dialog. That way your default settings won't be affected

Use the created INI file at the command-line like this:
$> mayo.exe file.stp --settings myconfig.ini --export file.stl

yvanblanchard, 2022-05-10:
Excellent !!
coudl you also tell me which (default tesselation) settings are used without any INI file defined in command-line ?
thank you

closed 2022-05-10 by HuguesDelorme