fougue/mayo #127: error in io_gmio_amf_writer.h  #include <gmio_amf/amf_document.h> no source

yilifeng, 2022-05-11:
[image]

[image]

Compile mayo on window, and the above error is reported. Did I forget to install something, but I did it according to the operation on the wiki. Has anyone encountered the same error and how to solve it, thank you very much

HuguesDelorme, 2022-05-11:
First time this kind of error is reported.
Usage of gmio library is enabled only if GMIO_ROOT option is specified in the qmake call.
What was the command-line you run for the qmake step ?

yilifeng, 2022-05-12:
> First time this kind of error is reported. Usage of gmio library is enabled only if GMIO_ROOT option is specified in the qmake call. What was the command-line you run for the qmake step ?
The compilation error of the missing file is as follows：
[image]

thank you for your reply
I downloaded gmio (https://github.com/fougue/gmio/releases/tag/v0.4.1) and added it to the environment, the compilation problem was solved, but an error was reported when I tested and exported the gltf format file, other functions are functioning normally.

Exporting gltf file reports the following error
[image]

HuguesDelorme, 2022-05-12:
It seems you built OpenCascade by yourself. I suggest you enable RapidJSON in OpenCascade by running CMake again or just download and install the official binaries for OpenCascade(see https://dev.opencascade.org/release)