fougue/mayo #129: Almost managed to build for aarch64

CheLoV, 2022-05-16:
Greetings!
Currently have OCCT 7.6.0. that is built from sources. Managed to build using qmake with command line according to your instruction. Though when i launch i get an error 

`./mayo: symbol lookup error: ./mayo: undefined symbol: _ZN35BinLDrivers_DocumentRetrievalDriver4ReadERK26TCollection_ExtendedStringRKN11opencascade6handleI12CDM_DocumentEERKNS4_I15CDM_ApplicationEERKNS4_I17PCDM_ReaderFilterEERK21Message_ProgressRange`

Any ideas? Clearly it refers to OCCT. But maybe also some compiler issues?

 Building or playing in qtcreator fails but there are plenty of other possibles reasons. I dont have experience with qt creator and possibly my environment is not tuned right. When i try to build with Qt i have multiple other errors.
```

/home/vizard/SOFT/Mayo/mayo/src/io_occ/io_occ.cpp:76: ??????: undefined reference to `Mayo::IO::OccGltfReader::createProperties(Mayo::PropertyGroup*)'
:-1: ??????: /home/vizard/SOFT/Mayo/build-mayo-Desktop-Debug/../mayo/src/io_occ/io_occ.cpp:78: undefined reference to `Mayo::IO::OccObjReader::createProperties(Mayo::PropertyGroup*)'
:-1: ??????: io_occ.o: in function `Mayo::IO::OccFactoryWriter::createProperties(Mayo::IO::Format, Mayo::PropertyGroup*) const':
/home/vizard/SOFT/Mayo/mayo/src/io_occ/io_occ.cpp:137: ??????: undefined reference to `Mayo::IO::OccGltfWriter::createProperties(Mayo::PropertyGroup*)'
:-1: ??????: /home/vizard/SOFT/Mayo/build-mayo-Desktop-Debug/../mayo/src/io_occ/io_occ.cpp:142: undefined reference to `Mayo::IO::OccObjWriter::createProperties(Mayo::PropertyGroup*)'
:-1: ??????: io_occ.o: in function `std::_MakeUniq<Mayo::IO::OccGltfReader>::__single_object std::make_unique<Mayo::IO::OccGltfReader>()':
/usr/include/c++/10/bits/unique_ptr.h:962: ??????: undefined reference to `Mayo::IO::OccGltfReader::OccGltfReader()'
:-1: ??????: io_occ.o: in function `std::_MakeUniq<Mayo::IO::OccObjReader>::__single_object std::make_unique<Mayo::IO::OccObjReader>()':
/usr/include/c++/10/bits/unique_ptr.h:962: ??????: undefined reference to `Mayo::IO::OccObjReader::OccObjReader()'
:-1: ??????: io_occ.o: in function `Mayo::IO::OccGltfWriter::OccGltfWriter()':
/home/vizard/SOFT/Mayo/mayo/src/io_occ/io_occ_gltf_writer.h:21: ??????: undefined reference to `vtable for Mayo::IO::OccGltfWriter'
:-1: ??????: /home/vizard/SOFT/Mayo/build-mayo-Desktop-Debug/../mayo/src/io_occ/io_occ_gltf_writer.h:21: undefined reference to `vtable for Mayo::IO::OccGltfWriter'
:-1: ??????: io_occ.o: in function `Mayo::IO::OccObjWriter::OccObjWriter()':
/home/vizard/SOFT/Mayo/mayo/src/io_occ/io_occ_obj_writer.h:20: ??????: undefined reference to `vtable for Mayo::IO::OccObjWriter'
:-1: ??????: /home/vizard/SOFT/Mayo/build-mayo-Desktop-Debug/../mayo/src/io_occ/io_occ_obj_writer.h:20: undefined reference to `vtable for Mayo::IO::OccObjWriter'
:-1: ??????: collect2: error: ld returned 1 exit status
:-1: ??????: [Makefile:647: mayo] Error 1
```

CheLoV, 2022-05-16:
I build on Alt Linux 10. With aarch64 processor.

HuguesDelorme, 2022-05-17:
Hello CheLoV

The first thing you could try is to run the DRAWEXE utility, it is probably there from the build you made. Have a look in the folder where are located the opencascade `.so` files.

Aside from that I would recommend to install "official" opencascade packages for your distro, it seems to provide v7.5.3
Then try to build Mayo using that version.

Report your progress here, good luck!

HuguesDelorme, 2022-05-17:
See https://packages.altlinux.org/en/p10/srpms/opencascade/

HuguesDelorme, 2022-05-17:
Relates to GitHub #100

CheLoV, 2022-05-17:
Greetings!
1) Drawexe works fine
2) we have tried the pachage from ALT Linux 7.5.3 but it only caused more issues. As far as we know the package actually have inner bugs and ours built from source is more stable (we also hae rapidjson there, draco, gles and opengl support, freeimage support)

```
ld: mesh_utils.o: в функции «Mayo::MeshUtils::allocateNormals(opencascade::handle<Poly_Triangulation> const&)»:
mesh_utils.cpp:(.text+0xa54): неопределённая ссылка на «Poly_Triangulation::AddNormals()»
ld: mesh_utils.o:(.data.rel.ro._ZTV20Standard_DomainError[_ZTV20Standard_DomainError]+0x40): неопределённая ссылка на «Standard_Failure::GetStackString() const»
ld: mesh_utils.o:(.data.rel.ro._ZTV20Standard_DomainError[_ZTV20Standard_DomainError]+0x48): неопределённая ссылка на «Standard_Failure::SetStackString(char const*)»
ld: mesh_utils.o:(.data.rel.ro._ZTV19Standard_OutOfRange[_ZTV19Standard_OutOfRange]+0x40): неопределённая ссылка на «Standard_Failure::GetStackString() const»
ld: mesh_utils.o:(.data.rel.ro._ZTV19Standard_OutOfRange[_ZTV19Standard_OutOfRange]+0x48): неопределённая ссылка на «Standard_Failure::SetStackString(char const*)»
ld: mesh_utils.o:(.data.rel.ro._ZTV21Standard_TypeMismatch[_ZTV21Standard_TypeMismatch]+0x40): неопределённая ссылка на «Standard_Failure::GetStackString() const»
ld: mesh_utils.o:(.data.rel.ro._ZTV21Standard_TypeMismatch[_ZTV21Standard_TypeMismatch]+0x48): неопределённая ссылка на «Standard_Failure::SetStackString(char const*)»
ld: xcaf.o:(.data.rel.ro._ZTV21Standard_NoSuchObject[_ZTV21Standard_NoSuchObject]+0x40): неопределённая ссылка на «Standard_Failure::GetStackString() const»
ld: xcaf.o:(.data.rel.ro._ZTV21Standard_NoSuchObject[_ZTV21Standard_NoSuchObject]+0x48): неопределённая ссылка на «Standard_Failure::SetStackString(char const*)»
ld: io_occ_base_mesh.o: в функции «Mayo::IO::OccBaseMeshReader::transfer(opencascade::handle<Mayo::Document>, Mayo::TaskProgress*)»:
io_occ_base_mesh.cpp:(.text+0x3a4): неопределённая ссылка на «RWMesh_CafReader::SetDocument(opencascade::handle<TDocStd_Document> const&)»
ld: io_occ_base_mesh.o:(.data.rel.ro._ZTV26Standard_ConstructionError[_ZTV26Standard_ConstructionError]+0x40): неопределённая ссылка на «Standard_Failure::GetStackString() const»
ld: io_occ_base_mesh.o:(.data.rel.ro._ZTV26Standard_ConstructionError[_ZTV26Standard_ConstructionError]+0x48): неопределённая ссылка на «Standard_Failure::SetStackString(char const*)»
ld: io_dxf.o:(.data.rel.ro._ZTV19Standard_RangeError[_ZTV19Standard_RangeError]+0x40): неопределённая ссылка на «Standard_Failure::GetStackString() const»
ld: io_dxf.o:(.data.rel.ro._ZTV19Standard_RangeError[_ZTV19Standard_RangeError]+0x48): неопределённая ссылка на «Standard_Failure::SetStackString(char const*)»
ld: io_dxf.o:(.data.rel.ro._ZTV21Standard_ProgramError[_ZTV21Standard_ProgramError]+0x40): неопределённая ссылка на «Standard_Failure::GetStackString() const»
ld: io_dxf.o:(.data.rel.ro._ZTV21Standard_ProgramError[_ZTV21Standard_ProgramError]+0x48): неопределённая ссылка на «Standard_Failure::SetStackString(char const*)»
ld: io_dxf.o:(.data.rel.ro._ZTV25BRepBuilderAPI_MakeVertex[_ZTV25BRepBuilderAPI_MakeVertex]+0x28): неопределённая ссылка на «BRepBuilderAPI_MakeShape::Build(Message_ProgressRange const&)»
ld: graphics_object_driver.o: в функции «Mayo::GraphicsMeshObjectDriver::createObject(TDF_Label const&) const»:
graphics_object_driver.cpp:(.text+0x16d4): неопределённая ссылка на «BRep_TFace::Triangulation(unsigned int) const»
ld: application.o: в функции «Mayo::Application::InitDocument(opencascade::handle<CDM_Document> const&) const»:
application.cpp:(.text+0x6ac): неопределённая ссылка на «TDocStd_Application::InitDocument(opencascade::handle<CDM_Document> const&) const»
ld: application.o: в функции «Mayo::Application::openDocument(std::filesystem::__cxx11::path const&, PCDM_ReaderStatus*)»:
application.cpp:(.text+0x2200): неопределённая ссылка на «TDocStd_Application::Open(TCollection_ExtendedString const&, opencascade::handle<TDocStd_Document>&, opencascade::handle<PCDM_ReaderFilter> const&, Message_ProgressRange const&)»
ld: application.o:(.data.rel.ro._ZTVN4Mayo8Document27FormatBinaryRetrievalDriverE[_ZTVN4Mayo8Document27FormatBinaryRetrievalDriverE]+0x30): неопределённая ссылка на «BinLDrivers_DocumentRetrievalDriver::Read(TCollection_ExtendedString const&, opencascade::handle<CDM_Document> const&, opencascade::handle<CDM_Application> const&, opencascade::handle<PCDM_ReaderFilter> const&, Message_ProgressRange const&)»
ld: application.o:(.data.rel.ro._ZTVN4Mayo8Document27FormatBinaryRetrievalDriverE[_ZTVN4Mayo8Document27FormatBinaryRetrievalDriverE]+0x38): неопределённая ссылка на «BinLDrivers_DocumentRetrievalDriver::Read(std::istream&, opencascade::handle<Storage_Data> const&, opencascade::handle<CDM_
```

3) I assume it may be related to qmake and some comliper issues. We encounteres it earlier in this sample project https://github.com/gkv311/occt-samples-qopenglwidget
Building with Qt and qmake also caused reference error, but building from cmakelists worked fine.
Is it possible to make a cmakelist file to test it out. Unfortunately don't have enough experience to make it from scratch on my own. Maybe you can assist with guidence?
Or making a cmake file is a difficult task?

HuguesDelorme, 2022-05-17:
Thanks for your feedback @CheLoV 
One thing to ensure for the QMake call is the path to OpenCascade libs.
With debian-based systems that is implicit and doesn't need to be specified.
That might be needed for Alt Linux though
Try to specify that path on command-line for qmake step:
````
$> qmake path/to/mayo.pro  CASCADE_INC_DIR=path/to/opencascade/includes  CASCADE_LIB_DIR=path/to/opencascade/libs
````

CheLoV, 2022-05-17:
I am not sure i can explain why, but i made a mistake and installed my custom built OCCT (7.6.0) above already installed 7.5.3 from ALT packages and mayo managed to built correctly and launched! That is a leap forward indeed.

I also managed to open a step file. but the performance in pretty low.
When i opened the same file with OCCT Draw i had around 10-15 FPS. Here i have 1 or less.

I also get a message in terminal TKOpenGl | Type: Other | ID: 0 | Severity: Medium | Message:
  OpenGl_Window::CreateWindow: window Visual is incomplete: no depth buffer, no stencil buffer

Any idea what that might be and what i might do to improve performance?

CheLoV, 2022-05-18:
The OCCT Samples Qopenhlwidget is a siple cube rendered, so it doesnt have issues (we are yet figuring out how to put a custom model there to test it out)

Will try out your advice, thanks!

CheLoV, 2022-05-26:
Greetings!
I accidently found out that OCCT Draw actually gives me also very low FPS. That is strange. 7.5.3 worked fine, 7.6.0 and later versions gives me a critical frame drop. The strange part that it drops when i just scale the window size.
I have a feeling that it happened due to 7.6.0. introduced ARM processor of M1 with ray tracing. Maybe that is the issue. Do you have an idea how to optimise graphic workflow of OCCT?

HuguesDelorme, 2022-05-30:
Hello @CheLoV 
Sorry no idea about that issue, you should report this into the OpenCascade forums(see https://dev.opencascade.org/forums)

closed 2022-05-30 by HuguesDelorme