fougue/mayo #132: Crash when close a document!

akjay, 2022-06-01:
I build the project in the way add all files to VS2019 project manually,add 3rdParty library，and it works,I build it successfully.But there is a problem I don't know why.When I close a document, the software carshed，so I check where bug happens, finnaly located that there is a code part in TDocStd_Application::Close,line 218(OCC7.5.0)

> aDoc->BeforeClose();  
> CDF_Application::Close(aDoc)

After the beforeClose called, the aDoc pointer while become a null pointer,so the next statement will crash.
Since we inherit TDocStd_Document and rewrite the function BeforeClose, so I step into this function find after the `delete guiDoc` called in file `gui_application.cpp`, the aDoc will be a nullptr,if I comment this `delete guiDoc`, everything will be fine

HuguesDelorme, 2022-06-01:
Hello akjay

Thanks for reporting that issue. Is that crash systematic?
I can't reproduce it but after looking the code I see the potential issue: `GuiDocument` might destroy the `Document` object.
Within `GuiApplication::onDocumentAboutToClose()` the reference counter of the `Document` shared pointer might become 0 and gets destroyed.
Statement `delete guiDoc;` should be replaced by `guiDoc->deleteLater();` so the `GuiDocument` gets safely destroyed when control returns to the event loop

HuguesDelorme, 2022-06-01:
@akjay 
Do you experience the crash when building Mayo with qmake/nmake as described in https://github.com/fougue/mayo/wiki/Build-instructions-for-Windows-MSVC ?

akjay, 2022-06-02:
> @akjay Do you experience the crash when building Mayo with qmake/nmake as described in https://github.com/fougue/mayo/wiki/Build-instructions-for-Windows-MSVC ?

Yes，there was no crash when use qmake/nmake to build.So it was a little bit strange.

HuguesDelorme, 2022-06-03:
@akjay
I've looked deeper at the code and actually the Document should not be destroyed because it's also owned by the parent `CDF_Application` object.
`Mayo::Application` inherits `TDocStd_Application` inherits `CDF_Application`
That latter class has a member of type `CDF_Directory` which stores `Document` objects.
`Mayo::Application::closeDocument()` calls `TDocStd_Application::Close()` which calls `CDF_Application::Close()` once `Document::BeforeClose()` has returned

Anyway feel free to attach here your VisualStudio files, I can try to reproduce the issue

HuguesDelorme, 2022-06-03:
@akjay
Make sure in your VS project you define the same build options as in mayo.pro
Especially those defined in opencascade.pri such as
`DEFINES += OCCT_HANDLE_NOCAST`
This might have a sensible impact

akjay, 2022-06-06:
> @akjay Make sure in your VS project you define the same build options as in mayo.pro Especially those defined in opencascade.pri such as `DEFINES += OCCT_HANDLE_NOCAST` This might have a sensible impact

Yeah，I forgot to define this macro，it works well and no crash with this macro defined

closed 2022-06-07 by HuguesDelorme