fougue/mayo #133: Qt5WinExtras dll dependency

yvanblanchard, 2022-06-02:
Hello,

I compiled your code on my computer, and everything works fine, but when sending the binaries and executable (including all necessary Qt dependencies DLLs) on another computer, it fails , saying that "Qt5WinExtras.dll" is missing. 
This file is not part of my runtime folder, and it works properly on my computer. Do you know this issue ?

HuguesDelorme, 2022-06-02:
You should find this DLL in the `bin/` folder of your Qt installation
Copy this DLL as well, it's used to report the global import/export progress in the Windows taskbar button of the Mayo application

yvanblanchard, 2022-06-02:
OK , thanks!
but how do you explain that at my side, when running the mayo exe without this DLL file in the folder, it works. 
The system succeeds at finding the DLL in my QT install bin folder ?

yvanblanchard, 2022-06-02:
I run the mayo.exe from a runtime folder, directly by double clicking on it (not from IDE).
In my PATh variable, I have %QTDIR%, so I guess this DLL was found automatically..

HuguesDelorme, 2022-06-02:
Then try to remove `%QTDIR%\bin` from your PATH variable and check again, you might miss other DLLs

yvanblanchard, 2022-06-02:
Ok thanks.
I also noticed that freeimage.dll file was missing.
Apart QT, do you list of other dependencies (and corresponding download link) ?
Thank you again.

yvanblanchard, 2022-06-02:
Thank you. I finally downloaded and compiled the missing dependencies (freetype and freeimage), and now it works.

closed 2022-06-02 by yvanblanchard