fougue/mayo #138: Change model material

akjay, 2022-06-29:
In some case I want to change a STEP model's material to get a better present with different light source,I tried to implement this function with code below   

> _//get the desired material type from user input or someway else_   
> auto materialType = Graphic3d_NameOfMaterial_***;    
> auto changeMaterial = [=] (GraphicsObjectPtr obj) {      
> Handle_XCAFPrs_AISObject aisObj = Handle_XCAFPrs_AISObject::Downcast(obj);   
> if(aisObj) aisObj->SetMaterial(materialType);};   
> auto guiDoc = ***;      //get current gui doc      
> auto roots = guiDoc->document()->modelTree().roots();   
> for (auto root : roots)    
> {   
> guiDoc->foreachGraphicsObject(root, changeMaterial);   
> }       
> guiDoc->graphicsScene()->redraw();      

But I found it's only useful for simply files, while assembly files this  code take no effect because the aisObj is nullptr and will not execute the if statement.

The object is created by `GraphicsShapeObjectDriver` and default material is `Graphic3d_NOM_PLASTER`,it is created with type `XCAFPrs_AISObject`, but in mapEntity function in class `GuiDocument`, assembly file object will be connect to a `AIS_ConnectedInteractive` object and push_back to `gfxEntity.vecObject`   
 
>        if (!docModelTree.nodeIsRoot(id)) {
>           Handle_AIS_ConnectedInteractive gfxInstance = new AIS_ConnectedInteractive;
>           gfxInstance->Connect(gfxProduct, XCaf::shapeAbsoluteLocation(docModelTree, id));
>            gfxInstance->SetDisplayMode(gfxProduct->DisplayMode());
>            gfxInstance->Attributes()->SetFaceBoundaryDraw(gfxProduct->Attributes()->FaceBoundaryDraw());
>            gfxInstance->SetOwner(gfxProduct->GetOwner());
>            gfxEntity.vecObject.push_back(GraphicsObjectPtr(gfxInstance));
>            if (XCaf::isShapeReference(docModelTree.nodeData(docModelTree.nodeParent(id))))
>                id = docModelTree.nodeParent(id);
>        }
>        else {
>            gfxEntity.vecObject.push_back(gfxProduct);
>        }   
   
So actually vecObject will store a `XCAFPrs_AISObject` pointer for simple file and `AIS_ConnectedInteractive` pointer for assembly file,and the latter one can not downcast to `XCAFPrs_AISObject` which case the problem when `changeMaterial ` function called.

I tried to call SetMetrial function direct without downcast which mease the lambda function like this:   
> auto changeMaterial = [=] (GraphicsObjectPtr obj) {   
> obj->SetMaterial(materialType);};   

Actually it called parent `PrsMgr_PresentablrObject` fucntion for assembly file and still take no effect.   
So I wonder how can I change a assembly file's material after it  has been created?

HuguesDelorme, 2022-06-29:
Hello @akjay
You should then try to apply `SetMaterial()` on the GraphicsObject pointed to by `AIS_ConnectedInteractive`:
````c++
auto changeMaterial = [=](GraphicsObjectPtr obj) {
    auto aisConn = Handle_AIS_ConnectedInteractive::Downcast(obj);
    auto aisXCaf = Handle_XCAFPrs_AISObject::Downcast(aisConn ? aisConn->ConnectedTo() : obj);
    if (aisXCaf)
        aisXCaf->SetMaterial(materialType);
};
````

akjay, 2022-06-29:
Thank for reply so quick. I will try what you say. @HuguesDelorme

akjay, 2022-06-29:
It works,thank you so much.

akjay, 2022-06-29:
> @akjay Notice you don't have to cast to `XCAFPrs_AISObject` as `SetMaterial()` is declared as a virtual function in the `PrsMgr_PresentableObject` root class. So your lambda could be simplified to:
> 
> ```c++
> auto changeMaterial = [=](GraphicsObjectPtr obj) {
>     auto aisConn = Handle_AIS_ConnectedInteractive::Downcast(obj);
>     const auto& aisObj = aisConn ? aisConn->ConnectedTo() : obj;
>     if (aisObj)
>         aisObj->SetMaterial(materialType);
> };
> ```

Can't agree more.

closed 2022-06-29 by akjay