fougue/mayo #141: new feature: Marking of Geometric tolerance

akjay, 2022-08-12:
Hi,is there a plan to develop a feature of marking geometric tolerance,like the CAD Assistant does.Or the OCCT supply a read to use function to do this?
[image]

HuguesDelorme, 2022-08-12:
CAD Assistant is using OpenCascade's "PMI Visualization Technology" commercial component.
That said OpenCascade has some free tools to support GD&T data when reading/writing STEP files. Main class `XCAFDoc_DimTolTool` even provides basic representations of GD&T data as BRep shapes.
Just displaying thoses shapes would be better than nothing.

akjay, 2022-08-12:
> CAD Assistant is using OpenCascade's "PMI Visualization Technology" commercial component. That said OpenCascade has some free tools to support GD&T data when reading/writing STEP files. Main class [...]

While,use OCAF tool,I can read GD&T data from STEP file, but how to display it?For dimensions, the library supply classes  under PrsDim folder to add an AIS_InteractiveObject created from selected shapes to show length,angular or other dimension types.

HuguesDelorme, 2022-08-12:
1. Call `XCAFDoc_DimTolTool::GetDimensionLabels()` to get the list of dimension of labels
2. For each dimension label retrieve the attached `XCAFDoc_Dimension` attribute
3. Call `XCAFDoc_Dimension::GetObject()` to retrieve `XCAFDimTolObjects_DimensionObject` instance
4. Retrieve the graphical presentation as a BRep shape by calling `XCAFDimTolObjects_DimensionObject::GetPresentation()`
5. The shape can be displayed with `AIS_Shape` directly

akjay, 2022-08-12:
Thank you, I just called the GetGDTPresentations function to get all TopoDS_Shape for the GD&T,the Display it out, it workd well,but can not move when drag,seems have mush work to do  for this.

HuguesDelorme, 2022-08-12:
If you want to move the dimensions 3D objects with the mouse then yes it requires some work.

akjay, 2022-08-16:
There is a question, if  I add some geometric tolerance information,how to compute the presentation manually？Dose occ support some function to do this,did not seen ant information about this in the user_guide?

akjay, 2022-08-19:
Is the presentation should be draw by myself，the occ did not provide a tool to draw this except pmi visualization commercial module？

HuguesDelorme, 2022-08-19:
Actually after adding some GD&T information a document, you would like to draw the represenation shape but it's not available(because it's new) ?

akjay, 2022-08-19:
Yes，for the STEP file imported in before，I can show presentations for these GD&Ts because they are part of ap242 file and maybe already drawn by other software，so the function can get TopoDS_Shape and show it.But if add a new definition，the presentation need to draw by self because occ did not provide a function to do this(maybe I didn't find it),if I save this file，only GD&T information can be stored in ap242 schema，no presentation since it has not been drawn.
And I even try define a GD&T through SolidWorks2021 and published use ap242，then imported in occ and try to read it's presentation，still failed with only part of the presentation shown like only an arrow but missing other parts.But if I use SolidWorks itself to read this file，it can be shown correctly，maybe each software has a different implement for the ap242 protocol？
But it seems that if I draw some presentation for the GD&T defined in occ and save it with ap242_dis schema，the open it with SolidWorks2021，it can be shown correctly.

HuguesDelorme, 2022-08-19:
I guess SolidWorks doesn't use the GD&Ts shape representations available in the STEP file but rather create its own representation/graphics.
This is surely what is also providing OpenCascade "PMI Visualization Technology" commercial component.

akjay, 2022-08-23:
Eh,I found a tool named [SFA ](https://dev.opencascade.org/project/step-file-analyzer)in opencascade  website told that can process STEP files.
And I tried with my model drawn by Solidworkd 2021,pictures below show the GD&T representation in different software.
1.Solidworks 2021 open a AP242 file with PMI enabled
[image]

2.CAD Assistant  with PMI commercial component [missing some infomation]
[image]

3.CAD Assistant without PMI commercial component [can not show ]
[image]

4.SFA
[image]

While the SFA show the GD&T infos same with Solidworks,so I guess they are should be read from AP242 files not draw by Solidworks itself, and this indicate that  when Solidworks  publish a AP242 file, the GD&T representation is well saved with standard ap242 protocol and can be readout by other software like OpenCASCADE(because SFA is based on OpenCASCADE),I try to read the SFA source code to find how it draw the GD&T representation but  it write with tcl language which is not my domain，and can not understant.

Do you have any idea？Or the SFA use other technology to achieve this?

Need your help.

Attach is my test STP file.
[part4.zip](https://github.com/fougue/mayo/files/9402137/part4.zip)

akjay, 2022-08-24:
Here is the source [code](https://github.com/usnistgov/SFA)

cz-source, 2022-10-19:
@akjay Can you share your code?  Using the OpenCascade for reading the GD&T and displaying it out, Thank you!

akjay, 2022-10-19:
> @akjay Can you share your code? Using the OpenCascade for reading the GD&T and displaying it out, Thank you!

Use this.
1.Call XCAFDoc_DimTolTool::GetDimensionLabels() to get the list of dimension labels
2.For each dimension label retrieve the attached XCAFDoc_Dimension attribute
3.Call XCAFDoc_Dimension::GetObject() to retrieve XCAFDimTolObjects_DimensionObject instance
4.Retrieve the graphical presentation as a BRep shape by calling XCAFDimTolObjects_DimensionObject::GetPresentation()
5.The shape can be displayed with AIS_Shape directly

cz-source, 2022-10-20:
> > @akjay 
Use this. 1.Call XCAFDoc_DimTolTool::GetDimensionLabels() to get the list of dimension labels 2.For each dimension label retrieve the attached XCAFDoc_Dimension attribute 3.Call XCAFDoc_Dimension::GetObject() to retrieve XCAFDimTolObjects_DimensionObject instance 4.Retrieve the graphical presentation as a BRep shape by calling XCAFDimTolObjects_DimensionObject::GetPresentation() 5.The shape can be displayed with AIS_Shape directly

Thank you. When I used PythonOCC for step 2, there was a problem. Maybe it's the version.

akjay, 2022-10-20:
I'm using C++, not know how to do this with PythonOCC.